robomechanics/quad-sdk #180: catkin_make meet redefinition error

littleggghost, 2021-12-06:
After  sudo ./setup.sh, I run catkin_make, but meet an error.
Could someone help me?

```
In file included from /usr/include/eigen3/Eigen/StdVector:23:0,
                 from /usr/local/include/rbdl/rbdl_math.h:37,
                 from /usr/local/include/rbdl/rbdl.h:11,
                 from /home/xxx/catkin_ws/src/quad-software/quad_utils/include/quad_utils/quad_kd.h:13,
                 from /home/xxx/catkin_ws/src/quad-software/quad_utils/include/quad_utils/math_utils.h:18,
                 from /home/xxx/catkin_ws/src/quad-software/global_body_planner/include/global_body_planner/planning_utils.h:17,
                 from /home/xxx/catkin_ws/src/quad-software/global_body_planner/src/planning_utils.cpp:1:
/usr/local/include/rbdl/rbdl_eigenmath.h:17:1: error: specialization of ‘std::vector<Eigen::Matrix<double, 2, 1> >’ after instantiation
 EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Eigen::Vector2d)
 ^
/usr/local/include/rbdl/rbdl_eigenmath.h:17:1: error: redefinition of ‘class std::vector<Eigen::Matrix<double, 2, 1> >’
In file included from /usr/include/c++/7/vector:64:0,
                 from /home/xxx/catkin_ws/src/quad-software/global_body_planner/include/global_body_planner/planning_utils.h:5,
                 from /home/xxx/catkin_ws/src/quad-software/global_body_planner/src/planning_utils.cpp:1:
/usr/include/c++/7/bits/stl_vector.h:216:11: note: previous definition of ‘class std::vector<Eigen::Matrix<double, 2, 1> >’
     class vector : protected _Vector_base<_Tp, _Alloc>
```

jcnorby, 2021-12-06:
We recently switched naming conventions and build systems which could cause redefinition errors. Try removing your build/ and devel/ folders then recompiling with `catkin build`, and let us know if that clears the issue.

littleggghost, 2021-12-07:
I tried with `catkin build`, but get the same error.

littleggghost, 2021-12-08:
Or could you tell me a branch/commit before switch naming? Then I can try to run the simulation. 
Thanks.

jcnorby, 2021-12-08:
Ah, we did modify RBDL slightly to clear these errors which is why there is a modified version in quad-software/external. Looks like your system is using the default one - you could either uninstall/remove those or copy our version into your install space.

littleggghost, 2021-12-08:
Very thanks for your help. I solved it by manually install rbdl-orb at `~/catkin_ws/src/quad-software/external/rbdl-orb` folder.

littleggghost, 2021-12-08:
By the way, [the wiki](https://robomechanics.github.io/spirit-software/) page seems not works.

jcnorby, 2021-12-08:
Just fixed, looks like that link didn't get updated when we changed the repo name a few day ago. However the Doxygen page is only going to help with the code API, not how to launch the simulator. You can find that information on the wiki [here](https://github.com/robomechanics/quad-software/wiki/4.-Gazebo-Simulator). I think I've finished updating the wiki pages to use the quad* name rather than spirit* but if you find any more spirit* anywhere, let us know.

closed 2021-12-08 by jcnorby

littleggghost, 2021-12-08:
Thanks so much!

yanhaoy, 2021-12-08:
> Ah, we did modify RBDL slightly to clear these errors which is why there is a modified version in quad-software/external. Looks like your system is using the default one - you could either uninstall/remove those or copy our version into your install space.

Yes, I modified RBDL by commenting out the lines 17-21 in rbdl_eigenmath.h as you can find in our external folder.