robomechanics/quad-sdk #192: Test performance in Noetic

jcnorby, 2022-01-28:


soldierofhell, 2022-05-28:
Hi @jcnorby , I'm interested in Noetic. Is there any work started?

soldierofhell, 2022-05-31:
Thanks. In the meantime I've managed to run the stack on Noetic (although I definitely haven't tested everything). I've got a question: what is the purpose of putting apt install ros-distro-package into setup_deps.sh (instead of using rosdep directly)?

jcnorby, 2022-06-06:
Glad to hear it seems mostly straightforward! And no particular reason other than that was the default way we had been building external packages so that's how they were added to the install scripts. We may have an additional update in the future which optimizes those setup scripts.

ardalantj, 2022-06-10:
@soldierofhell Can this issue be closed?

ardalantj, 2022-06-16:
Will close this issue due to inactivity. If there is a problem, open a new issue

closed 2022-06-16 by ardalantj