robomechanics/quad-sdk #204: Improve local planner solve when yaw wraps

jcnorby, 2022-02-20:
Right now if a twist input is commanded to spin in a circle, we get a much slower solve when the robot crosses from yaw = -pi to yaw = pi. Short term solution would be to update the warm start to unwrap yaw if it crosses that threshold. Longer term solution would be to use quaternions or some other representation.

yanhaoy, 2022-03-02:
Will try to wrap yaw in warmstart now, not sure will it affect the multipliers. Using a quaternion in direct collocation requires an extra slack variable to keep it as a unit vector.

dhajnes, 2023-09-13:
Hi, any news on this from the dev's side? It's a year and a half old problem that has allegedly "a pretty easy solution, we just haven't gotten around to it yet" citing #345. @jcnorby

ardalantj, 2023-09-13:
Hi @dhajnes, 
Our team is currently working on other higher priority like state estimation. We may or may not get to this in the near future.

jcnorby, 2023-09-17:
@dhajnes sorry it took so long to get to this! Finally found a time and opened a PR (#400) that resolves this issue. Feel free to merge into your branch, although its pretty self contained so should be merged with devel pretty soon.

Thanks for your interest!

closed 2023-11-30 by jcnorby