robomechanics/quad-sdk #283: The branch which can be used?

crankler, 2022-05-27:
Hi, it's a great work from your team,  
but there have many branch and I wonder whicn I can use for the contact detection and proprioceptive terrain estimation as your paper said hare?
>Proprioception and Tail Control Enable Extreme Terrain Traversal by Quadruped Robots

yanhaoy, 2022-05-30:
`add-tail-controller` contains most of the paper, but it's still a work in progress. There is a branch called `devel-terrain-estimation` that has basic terrain estimation stuff.

yanhaoy, 2022-05-30:
Thank you for your attention. It is ongoing work and is not part of the release. But there is a branch called "devel-terrain-estimation" which has basic terrain estimation stuff.

yanhaoy, 2022-05-30:
Thank you for your attention. It is ongoing work and is not part of the release. But there is a branch called `devel-terrain-estimation` which has basic terrain estimation stuff.

crankler, 2022-05-31:
Yes, I will check `Quad Utils pkg` and  `devel-terrain-estimation` branch in these days, thank you for your reply here.

closed 2022-05-31 by crankler