robomechanics/quad-sdk #288: nmpc solving fail

ZhangLiang-Cencen, 2022-06-07:
when I input roslaunch quad_utils quad_plan.launch reference:=twist logging:=true, the terminal shows "nmpc solving fail“.

prakharg01, 2022-06-07:
> when I input roslaunch quad_utils quad_plan.launch reference:=twist logging:=true, the terminal shows "nmpc solving fail“.

I'm facing a similar issue. I have installed HSL solver

ZhangLiang-Cencen, 2022-06-08:
Thanks for your answers. I started with ubuntu 20.04 and ros noetic, then  I solved this problem by using ubuntu 18 and ros melodic. And I refer to this article [https://zhuanlan.zhihu.com/p/522218252](url)

ZhangLiang-Cencen, 2022-06-08:
You may try this quad_sdk (someone else deals with it) [https://pan.baidu.com/s/1rVzICOEnplKmRzBL6bVuEQ 
c31w ](url)

littleggghost, 2022-06-08:
> You may try this quad_sdk (someone else deals with it) [https://pan.baidu.com/s/1rVzICOEnplKmRzBL6bVuEQ c31w ](url)

Hi @ZhangLiang-Cencen 
This link fails. Could you commit this link as a branch?

ZhangLiang-Cencen, 2022-06-08:
You may copy it to browser window. I just tested it.

jcnorby, 2022-06-08:
Sounds like there is mixed success here (and as mentioned in another thread the software is currently only tested in Melodic although Noetic support is increasing in priority). I'll chime in with some additional explanations and suggestions in case folks are still seeing this.

The NMPC solving fail warning indicates that the NLP solver (IPOPT) is for whatever unable to find a feasible solution to the optimal control problem in the given time duration (this is needed to compute the body and GRF trajectories). In a properly installed system with reasonable computational capabilities this shouldn't occur - for reference typical solve times are currently around 2-3ms on a desktop and closer to 10ms on a TX2.

Possible solutions/debugging approaches:
- Turn on IPOPT printing to see what the specific issue is (change the print_level on [this line](https://github.com/robomechanics/quad-sdk/blob/89154df18e98162249f38301b669df27ee595220/nmpc_controller/src/nmpc_controller.cpp#L212) to 5)
- Make sure HSL is installed - these solvers are much more efficient than the default. You can confirm that these are being used correctly by checking the IPOPT printout, it should say that the linear solver is ma57 and not mumps.
- If the IPOPT printout indicates that the maximum clock time was exceeded, you could give the solver more time by following the directions [here](https://github.com/robomechanics/quad-sdk/wiki/2.-Using-the-Software#common-roslaunch-calls).
- If the IPOPT printout indicates that something besides maximum time caused it to exit, there is likely something wrong with the install. Please post the result here so we can take a look.

jcnorby, 2022-06-09:
Quick clarification - did you put those files into the ipopt folder prior to calling the setup script? That script is responsible for actually building the source code, so if it is called before copying the HSL files then it would fail (if that is the case you can just re-run the setup script).

jcnorby, 2022-06-09:
Also make sure that `sudo apt update` and the setup script run smoothly, if they run into errors things may still build but if HSL isn't installed correctly then this error will occur (a note about this will be added to the installation instructions).

prakharg01, 2022-06-09:
Yes, I had added those files before calling the setup script.

Yeah, I have made sure both work, I have to check which exact version of HSL I installed. 
Thanks

prakharg01, 2022-06-09:
It's something to do with shared libraries and linking, Should I make a symbolic link or does the setup script do that?

yanhaoy, 2022-06-09:
> > * printout indi
> 
> Thanks for the detailed instructions!
> 
> After enabling printing, it looks like I have errors loading the shared library:-
> 
> Exception of type: DYNAMIC_LIBRARY_FAILURE in file "/QuadSdk/catkin_ws/src/quad-sdk/external/ipopt/coinbrew/Ipopt/src/Common/IpLibraryLoader.cpp" at line 67: Exception message: libhsl.so: cannot open shared object file: No such file or directory
> 
> EXIT: Library loading failure.
> 
> I have downloaded the correct version of HSL for IPOPT and placed it in the right directory as well. There are no build issues as well. Any help will be beneficial.
> 
> Thanks, Prakhar Goel.

Building or running codes should not depend on any files under external/ipopt/ after successfully running setup.sh as it installs all compiled libraries to /usr/local. For example, even if I clear the path external/ipopt/, I can still build and run the codes. Could you check if there's any error message pop up when you run setup.sh? Thanks!

prakharg01, 2022-06-10:
There is no error

prakharg01, 2022-06-10:
It's not able to find the shared library, does that mean it's something to do with the installation of hsl solver?

jcnorby, 2022-06-10:
Just to make sure the HSL files are in the right place - is quad-sdk/external/ipopt/coinhsl/libcoinhsl.sym a valid path?

prakharg01, 2022-06-10:
Yes they exist on the correct path 

After updating my .bashrc with correct settings for dynamic library loaders, I am getting this error now :- Exception message: /usr/local/lib/libhsl.so: undefined symbol: MA57AD

yanhaoy, 2022-06-10:
I believe the installed library is called libcoinhsl.so (at least on my machine), you can try setting up a symbolic link like [https://github.com/casadi/casadi/issues/2769](url). But I guess there are still some problems during the installation process, maybe the HSL version and the ThirdParty-HSL repo do not match.

jcnorby, 2022-06-10:
Okay, we may have found the issue (thanks @yanhaoy) - our NMPC controller looks for the MA57 solver, which is available with the Academic License of IPOPT but not with the free version. If folks have the free version, IPOPT will not find this solver and will default to mumps, which isn't fast enough.

Luckily, the MA27 solver is included in the free version, and seems to give nearly identical performance so we will just default to using that solver. We've added this change to an upcoming PR which should hopefully find its way into main by the end of the weekend. If folks want to try it out before then, either check out [this branch](https://github.com/robomechanics/quad-sdk/tree/melodic_devel_nmpc_refactor), or change the line in quad-sdk/nmpc_controller/src/nmpc_controller.cpp from
```
app_->Options()->SetStringValue("linear_solver", "ma57");
```
to
```
app_->Options()->SetStringValue("linear_solver", "ma27");
```
Let us know if this solves the issue!

closed 2022-06-13 by jrenaf