robomechanics/quad-sdk #291: global planner fail

ariomaleeq, 2022-06-09:
Hello, I compiled everything and ran the test. nothing failed. However, when I tried running the global planner, it always shutdown with the following error after switching to refinement mode:

[ INFO] [1654770393.108580679, 14.868000000]: In reset mode
[ INFO] [1654770393.110130506, 14.869000000]: Switching to refinement mode
munmap_chunk(): invalid pointer
[robot_1/global_body_planner-1] process has died [pid 32246, exit code -6, cmd /home/ario/quad_sdk_ws/devel/lib/global_body_planner/global_body_planner_node start_state:=state/ground_truth goal_state:=clicked_point __name:=global_body_planner __log:=/home/ario/.ros/log/91fd48aa-e7de-11ec-bdaa-d8f8836e6f86/robot_1-global_body_planner-1.log].
log file: /home/ario/.ros/log/91fd48aa-e7de-11ec-bdaa-d8f8836e6f86/robot_1-global_body_planner-1*.log

Thank you in advance

jcnorby, 2022-06-09:
Hey there! Just to clarify - was the 'test' that passed the unit tests? Or were you able to track a twist reference with:
```
roslaunch quad_utils quad_plan.launch reference:=twist logging:=true
rosrun teleop_twist_keyboard teleop_twist_keyboard.py cmd_vel:=/robot_1/cmd_vel
```

ariomaleeq, 2022-06-09:
the test is the catkin test that checks each package, I think that's the unit test. Using the twist reference with keyboard teleop is fine for me, but when I change the reference to gbpl, this error happened. Thanks for answering so quick!

error occurs when I do:
roslaunch quad_utils quad_plan.launch reference:=gbpl logging:=true

jcnorby, 2022-06-09:
Hmm, I'm not able to reproduce that error. Before it switches to refinement mode does it give a printout with some data like solve time, vertices, etc? That would indicate if it found a valid plan or not.

jcnorby, 2022-06-09:
You could also try compiling in Debug mode (`catkin build --cmake-args -DCMAKE_BUILD_TYPE=Debug`) to see if the printout becomes any more informative. Looks like some sort of segfaulting issue - sometimes debug mode introduced additional checks that can give more info. If you have the setup you could also try running it within a debugger to locate the exact issue, but without being able to recreate it locally I can't determine exactly what is causing the error, sorry!

ariomaleeq, 2022-06-10:
Hello thanks again for the quick response! That build option solved it for some reason, however, a new issue emerged, the robot started jumping around instead of following the line and the nmpc solver fails. However, this is not a part of this issue so I'm going to close this one. Thanks again for your help!

jcnorby, 2022-06-10:
I should warn that compiling in Debug results in terrible performance (map queries and NLP solves get MUCH slower), but can be useful for finding bugs. If you switch back to Release is the problem resolved?

ariomaleeq, 2022-06-12:
It failed again. There were multiple compilation warning when building the robot_driver package. One of which is:

warning: taking address of packed member of ‘__mavlink_local_position_ned_cov_t’ may result in an unaligned pointer value

also,
warning: taking address of packed member of ‘__mavlink_attitude_target_t’ may result in an unaligned pointer value [-Waddress-of-packed-member]

All of them were the same type of warnings about taking the address of packed members. I read online that this might cause memory alignment issues in some computer architecture. Is this possible that this is the issue with my computer?

jcnorby, 2022-06-12:
Hmm, I think that may be the case, these seem more like compiler/memory issues (especially since it seems that the errors are in inconsistent packages). I'm going to close this for now since I don't think the issue is with the global planner - perhaps try on a clean system or from a virtual machine.

closed 2022-06-12 by jcnorby