robomechanics/quad-sdk #295: Global body planner package running into failures (Building has no errors)

prakharg01, 2022-06-13:
Thanks for the continuous support!

After solving the local planner issues, running catkin run_tests results in failed tests for global planner

Total 8 tests failed:-

1) Expected: (avg_duration) <= (5e-6), actual: 3.06846e-05 vs 5e-06
[  FAILED  ] GlobalBodyPlannerTestFixture.testIsValidStateTime (357 ms)
[ RUN      ] GlobalBodyPlannerTestFixture.testGetRandomLeapActionTime

2) Expected: (avg_duration) <= (1e-4), actual: 0.00160519 vs 0.0001
[  FAILED  ] GlobalBodyPlannerTestFixture.testIsValidStateActionPairTime (2808 ms)

3) Actual: false
Expected: true
[  FAILED  ] GlobalBodyPlannerTestFixture.testValidStateActionPairRate (2772 ms)

4-8)
[  FAILED  ] 4 tests, listed below:
[  FAILED  ] GlobalBodyPlannerTestFixture.testIsValidStateTime
[  FAILED  ] GlobalBodyPlannerTestFixture.testGetRandomLeapActionTime
[  FAILED  ] GlobalBodyPlannerTestFixture.testIsValidStateActionPairTime
[  FAILED  ] GlobalBodyPlannerTestFixture.testValidStateActionPairRate

Any help will be beneficial,
Thanks,
Prakhar Goel

prakharg01, 2022-06-13:
I was using debug before, so building it in release mode doesn't give any failures.

But running the global planner using : - roslaunch quad_utils quad_plan.launch reference:=gbpl logging:=true gives an error

[robot_1/global_body_planner-1] process has died [pid 27520, exit code -6, cmd /home/prakhargoel/Desktop/Mitacs/QuadSdk/catkin_ws/devel/lib/global_body_planner/global_body_planner_node start_state:=state/ground_truth goal_state:=clicked_point __name:=global_body_planner __log:=/home/prakhargoel/.ros/log/add1c446-eb37-11ec-b33c-3cf01140c1cb/robot_1-global_body_planner-1.log].
log file: /home/prakhargoel/.ros/log/add1c446-eb37-11ec-b33c-3cf01140c1cb/robot_1-global_body_planner-1*.log

prakharg01, 2022-06-13:
I'm able to move the robot using teleop without any errors.

prakharg01, 2022-06-13:
Solve time: 0.00321668 s
Vertices generated: 6
Path length: 5.02697 m
Path duration: 10.053 s

[ INFO] [1655138528.430044797, 14.168000000]: In reset mode
[ INFO] [1655138528.432825660, 14.169000000]: Switching to refinement mode
free(): double free detected in tcache 2
After this it fails

jrenaf, 2022-06-13:
Running teleop does not require using global planner so it wouldn't give you related errors. From the error it seems the global planner has already figured out the valid solution but has the problem publishing it.

jcnorby, 2022-06-14:
I can shed some light on the test failures - we have some unit tests which require a certain level of computation efficiency of the underlying function calls. Those will fail if in Debug mode or on a particularly slow platform, but you can ignore them.

That said, debug mode should help you determine where the actual error is coming from, although if global planner node is crashing then it is likely somewhere internal to that library. Any information on what file or line is causing the failure would be useful (you can get this either with a debugger or with well-placed print statements).

prakharg01, 2022-06-14:
Oh okay, Thanks a lot! I will definitely try to locate the Ros error.

prakharg01, 2022-06-14:
After it switches to refinement mode, it shows that message. Does it mean it's something to do with memory?

"
[ INFO] [1655220178.130017436, 74.472000000]: Switching to refinement mode
free(): double free detected in tcache 2"

jcnorby, 2022-06-14:
That is certainly the symptom that throws the error but unlikely to be the root cause, it looks more like a segfault (which is why a debugger would be the best course of action).

prakharg01, 2022-06-14:
Okay, the error code is -4. The SIGILL signal is raised when an attempt is made to execute an invalid, privileged, or ill-formed instruction. SIGILL is usually caused by a program error that overlays code with data or by a call to a function that is not linked to the program load module.

jrenaf, 2022-06-14:
Yeah it looks like a memory linking issue. Can you use print statement or debugger to locate which line of code causes the issue?

prakharg01, 2022-06-15:
Can you suggest which file I should start debugging? I find no problems in the global_body_panner node.

prakharg01, 2022-06-15:
RUnning it in debug, 
After few solutions it ended up with this message:-
[ WARN] [1655308419.384749504, 7.286000000]: Planner was unable to make any progress, start state likely trapped
[ INFO] [1655308420.722279044, 7.741000000]: Switching to refinement mode

prakharg01, 2022-06-15:
The Goal pose can be changed in rviz or there is a fixed parameter to update in some launch file?

prakharg01, 2022-06-15:
> Running it in debug, After a few solutions, it ended up with this message:- [ WARN] [1655308419.384749504, 7.286000000]: Planner was unable to make any progress, start state likely trapped [ INFO] [1655308420.722279044, 7.741000000]: Switching to refinement mode

This was for the default goal state (5,0). Does that mean a global planner isn't able to find a solution?

prakharg01, 2022-06-15:
Yes as informed earlier, with twist commands the robot is walking on many terrains. The robot stands sucessfully. When running the global planner with a default goal (5,0) on flat terrain :-

ROS_MASTER_URI=http://localhost:11311

process[robot_1/global_body_planner-1]: started with pid [22992]
process[robot_1/local_planner-2]: started with pid [22993]
process[robot_1/rosbag_record-3]: started with pid [22994]
process[robot_1/rosbag_record_archive-4]: started with pid [22995]
process[robot_1/plan_publisher-5]: started with pid [22996]
[ INFO] [1655314392.520151306, 46.379000000]: GBP has state and map information
[ INFO] [1655314392.539085729, 46.383000000]: In reset mode
complexity_schedule = 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[ INFO] [1655314392.738179091, 46.427000000]: valid and shorter or previous plan not valid
Solve time: 0.186177 s
Vertices generated: 6
Path length: 3.7654 m
Path duration: 7.52545 s

[ WARN] [1655314397.225284166, 47.537000000]: **Planner was unable to make any progress, start state likely trapped**
[ INFO] [1655314400.812556352, 48.412000000]: Switching to refinement mode
[robot_1/global_body_planner-1] process has died [pid 22992, exit code -4, cmd /home/prakhargoel/Desktop/Mitacs/QuadSdk/catkin_ws/devel/lib/global_body_planner/global_body_planner_node start_state:=state/ground_truth goal_state:=clicked_point __name:=global_body_planner __log:=/home/prakhargoel/.ros/log/c9456daa-ecd0-11ec-bbd4-3cf01140c1cb/robot_1-global_body_planner-1.log].
log file: /home/prakhargoel/.ros/log/c9456daa-ecd0-11ec-bbd4-3cf01140c1cb/robot_1-global_body_planner-1*.log

jrenaf, 2022-06-16:
Ok, from the output, it seems like the first time running global planner succeeds. Could you check the param num_calls value in global_boduy_planner.yaml? If it is one, then it means something triggers the global planner runs again where at this time the global planner cannot get a valid result. Global planner is triggered again usually means the robot significantly deviates from the original plan. Another way to check this is to rostopic echo /robot_1/topics/global_plan and see whether it has the first plan published.

jcnorby, 2022-06-16:
I'm out for the next week or so but can briefly chime in with a debugging tip - if running the planner in debug mode, increase the max_planning_time parameter so something like a few seconds (you could also increase the startup delay to the same value). This allows the planner to take more time to find a feasible solution - usually the default value is fine but if compiling in debug mode the default is likely too fast.

prakharg01, 2022-06-16:
> Ok, from the output, it seems like the first time running global planner succeeds. Could you check the param num_calls value in global_boduy_planner.yaml? If it is one, then it means something triggers the global planner runs again where at this time the global planner cannot get a valid result. Global planner is triggered again usually because the robot significantly deviates from the original plan. Another way to check this is to rostopic echo /robot_1/topics/global_plan to see whether it has the first global plan published.

Okay, thanks. Looks like the plan is not getting published for some reason. Any possible reason you can think of ?

WARNING: topic [/robot_1/topics/global_plan] does not appear to be published yet

jrenaf, 2022-06-16:
One more way is to uncomment line 26-27 inside planning_utils.h
// #define DEBUG_REFINE_STATE
// #define DEBUG_INVALID_STATE
By doing this, it can gives you a bunch of more detailed information on what is going wrong.

jrenaf, 2022-06-16:
Ok, it seems like the global planner doesn't have time to figure out a feasible plan, try increase max_planning_time in global_body_planner.yaml from 0.20 to 1.0.

prakharg01, 2022-06-16:
I tried increasing startup time and max_planning_time. No major change is observed.

jrenaf, 2022-06-16:
Keep the max_planning_time to be few seconds while commenting out the debug output in planning_utils.h and try again?

prakharg01, 2022-06-16:
Yeah did that, it still doesn't seem to converge.

Solve time: 0.160707 s
Vertices generated: 6
Path length: 5.02893 m
Path duration: 10.0096 s

complexity_schedule = 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[ INFO] [1655394829.175067703, 9.029000000]: In reset mode
[ INFO] [1655394836.269547945, 11.082000000]: In reset mode
[ INFO] [1655394840.302016716, 12.223000000]: Switching to refinement mode
and then process died

prakharg01, 2022-06-16:
Hello, Is there anything else to try or can you suggest any method cause when I'm using the deprecated version of the global planner, I'm able to get solutions really quick. :- https://github.com/jcnorby/global_body_planner. I feel its something to do with the overall integration in general and my setup.

jcnorby, 2022-06-16:
You are correct - those printouts indicate that the planner found a valid plan, but something crashed between that and publishing the plan (so you can leave those debugging lines commented out).

I doubt this will solve it but from the printouts it looks like you are using a slightly old version of the code (there have been a few changes in the last few days), I'd recommend updating to the newest main.

We still need to figure out where in the code this is occurring for you. In lieu of using a debugger, add print statements to the spin function of the global_body_planner class to determine which subfunction is failing, then repeat that process in the subfunction until you determine which line is causing it to crash.

NmBoyd, 2022-06-17:
I've been working off both main and devel with their most recent versions. Both have this same issue with global_body_planner for me.

prakharg01, 2022-06-17:
Okay, Thanks a lot. 

I was with the latest changes on main. I will try debugging further and keep this thread updated as some other people are also facing similar issues I believe.

prakharg01, 2022-06-17:
> I've been working off both main and devel with their most recent versions. Both have this same issue with global_body_planner for me.

Hi! Let me know if anything solves the issue

prakharg01, 2022-06-17:
> I've been working off both main and devel with their most recent versions. Both have this same issue with global_body_planner for me.

When you are in debug mode, are you getting failed tests after running the catkin_tests?

prakharg01, 2022-06-21:
Hi!

I am unable to figure out what causes the issue, is there any other thing that I can try?

Thanks!

prakharg01, 2022-06-21:
Hi, Its something to do with this function :- void GlobalBodyPlanner::publishCurrentPlan()

There are 3 conditions that are checked, and for some reason, the message to publish is never created and it is stuck in these conditions, even if the path is valid.

prakharg01, 2022-06-22:
I have commented out the line :- _current_plan_.setPublishedTimestamp(ros::Time::now());_
It finds a path, I can visualize the red path in rviz as well, but the NMPC fails for some reason:-

found valid plan
[ WARN] [1655923323.488762071, 33.328000000]: NMPC solving fail
found valid plan
[ WARN] [1655923323.615151404, 33.361000000]: NMPC solving fail
[ WARN] [1655923323.739059278, 33.391000000]: NMPC solving fail
found valid plan
[ WARN] [1655923323.778932163, 33.401000000]: Start is sufficiently close to goal, exiting
[ WARN] [1655923323.867511179, 33.423000000]: NMPC solving fail
found valid plan
[ WARN] [1655923323.993851230, 33.456000000]: NMPC solving fail
[ WARN] [1655923324.116495899, 33.487000000]: NMPC solving fail
found valid plan
[ WARN] [1655923324.245972782, 33.526000000]: NMPC solving fail
found valid plan
[ WARN] [1655923324.378389956, 33.562000000]: NMPC solving fail
[ WARN] [1655923324.504381709, 33.594000000]: NMPC solving fail

Note: Found valid plan is a print statement I added for debugging purposes.

prakharg01, 2022-06-23:
Hey!

It solves the issue! Thanks a lot for the continuous support, I can now visualize the plan and the robot follows it with the NMPC controller also. There are some issues with leap tracking and it stops tracking the plan after some time(the plan publisher fails),  which I will tune parameters/try fixing further, but I can close this issue now:)

Hope it helps everyone!

closed 2022-06-23 by prakharg01

jrenaf, 2022-06-23:
That's a nice catch Prakhar, we will fix this bug in devel and main branch.