robomechanics/quad-sdk #296: Loading different robot types (e.g. a1)

AndrewZheng-1011, 2022-06-13:
First of all, I just want to say just how amazing and complete this software development kit is. I would like to thank you for providing such an end-to-end open-source development kit to the community. It really helps people without the right infrastructure to develop quadruped algorithms. Regarding the issue I am having with...
<br />

**Problem**
One of the things that I noticed that I couldn't get to work is that when I tried to run the unitree a1 quadruped, the rviz and gazebo model would consistently be loading spirit using the following command:

```
roslaunch quad_utils quad_gazebo.launch robot_type:="a1" gui:=true
```

<br />

**Issue**
Spirit model would consistently be loaded.

<br />

**Partial Solution**
The solution I found to this, though, was adding the "robot_type" argument to pass into the quad_visualization.launch file through the quad_gazebo.launch file. However, the problem quickly after that is that the model of the file is not properly exported/setup right for it to be properly found through the uri (e.g. spirit.sdf uri filepaths are model://a1_description/meshes/...). The error is shown below:

```
[Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [model://a1_description/meshes/hip.dae]
[Wrn] [SystemPaths.cc:464] File or path does not exist [""] [model://a1_description/meshes/hip.dae]
[Err] [Visual.cc:3046] No mesh specified
[Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [model://a1_description/meshes/hip.dae]
[Wrn] [SystemPaths.cc:464] File or path does not exist [""] [model://a1_description/meshes/hip.dae]
[Err] [Visual.cc:3046] No mesh specified
.
.
.
```
I believe this is due to the a1_description not being explicitly defined within the gazebo model path. Currently, the solution I have implemented is explicitly adding the quad_simulator folder into the gazebo model path through the quad_simulator/setup_deps.sh script. However, this is more of a bandaid solution because other minor issues rises as gazebo wants a nice structure as seen in the [link](https://classic.gazebosim.org/tutorials?tut=model_structure&cat=build_robot). I believe slightly changing the folder structure may be more beneficial and cleaner if we want to add any future models in, but I am not sure how this change will affect other parts of the code. I wanted to run this by to make sure if this was a known issue or not.

jcnorby, 2022-06-14:
Hey there, thanks for the kind words! With respect to that error, can you see if the devel branch fixes this issue for you? There was a bug with parameter loading that we resolved recently with the behavior you described (defaulting to spirit). There's a PR open right now to merge this to main (#294) which is awaiting review.

AndrewZheng-1011, 2022-06-14:
@lnotspotl, I switched to the devel branch and and was able to successfully load the a1 gazebo model. The process I took was:
```
cd ~/<your_catkin_ws>/src/quad-sdk/
./setup.sh
source ~/.bashrc
```
This should update your gazebo model path that the setup_deps.sh file wrote into the bashrc file. The expected results are shown below with a loaded a1 model in rviz and a loaded a1 model (with a green tint to the base) in gazebo. 
[image]

[image]

However, the terminal still outputs some error when running
```
roslaunch quad_utils quad_gazebo.launch gui:=true robot_type:=a1
```
where the terminal with gazebo gives minor error that doesn't affect the actual process:
```
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/andrewzheng1011/quad_sdk_ws/src/quad-sdk/quad_simulator/spirit_description/materials"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/andrewzheng1011/quad_sdk_ws/src/quad-sdk/quad_simulator/spirit_description/meshes"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/andrewzheng1011/quad_sdk_ws/src/quad-sdk/quad_simulator/spirit_description/urdf"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/andrewzheng1011/quad_sdk_ws/src/quad-sdk/quad_simulator/a1_description"
.
.
.
```
This error is more so the fact that gazebo doesn't like the folder structure of the exported paths where the ideal folder structure one stated in the link I posted on my first comment. This is definitely more of a minor error that doesn't really affect any of the operations as I am able to load the model. @jcnorby, I am not sure how much you want to rearrange these model files for the sake of cleanness and having no error pop up in the terminal. I would say that in the future, I would suggest that we have a file structure such as this:
```
quad_simulator
├── robot_models                   # Robot model files
│   ├── spirit_description
│    │   ├── model.config
│    │   ├── model.sdf
│    │   ├── meshes
│   ├── a1_description
│   ├── other                         # Accessory (e.g. sensors)
```

This way, it is clean in adding model paths into the ./bashrc file. and clean to write the export environment gazebo model path command in the quad_simulator/setup_sh.deps.sh. Additionally, adding more models should be a cleaner process. I can spend some time to see if rearranging these folders and editing the uri paths in the dae file will still cause the rest of the program to run smoothly if the developers of this repository want to go down this route.

AndrewZheng-1011, 2022-06-14:
Another issue does arise when running the setup.sh script. I am not sure if this is the right spot to post this (I am still fairly new to github) or if I should post this as a different issue. When switching to different branches, I recopied the coinhsl path into /external/ipopt/coinhsl and ran the quad_sdk/setup.sh script. Here, I run into issue regarding ipopt etc. As the script is running, I run into some error such as the following:
```
##################################################
### Building ThirdParty/HSL 2.2.1 
##################################################
.
.
.
/path/to/quad-sdk/external/ipopt/coinbrew/ThirdParty/HSL/coinhsl/hsl_mc68i_ciface.f90:3:7:
    use hsl_mc68_integer, only: &
       1
Fatal Error: Can't open module file ‘hsl_mc68_integer.mod’ for reading at (1): No such file or directory
compilation terminated.
make[1]: *** [coinhsl/hsl_mc68/C/hsl_mc68i_ciface.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
.
.
.
```
This happens for many other hsl files during the shell script process. However, I am still able to run the planner just fine. I just want to verify if these are expected errors that you guys face @jcnorby or there is a different procedure in place for switching between different branches and recopying the coinhsl file into the external/ipopt/ directory.

jrenaf, 2022-06-14:
Hi, that's a good suggestion to reorganize the structure of the description files. We will fix this bug in the next merge. @AndrewZheng-1011 For your second problem about coinhsl, could you open a new issuse please?

yanhaoy, 2022-06-15:
We don't have this HSL error in our machine.  Can you check if the path structure is correct, like quad-sdk/external/ipopt/coinhsl/ where all the ma* files and folders are there?  Thanks!

AndrewZheng-1011, 2022-06-15:
My path structure looks something like this:
```
quad-sdk
├── external
│   ├── hsl_ma77
│   ├── hsl_ma86
│   │   ├── C/
│   │   ├── hsl_ma86d.f90
│   │   ├── hsl_ma86d.lo
│   │   ├── hsl_ma86ss.f90
│   │   ├── hsl_m86s.lo
│   ├── hsl_ma96
│   ├── hsl_ma68
│   ├── ma27
│   ├── ma28
│   ├── ma57
└── ...

```

I had no error when initially setting up ipopt and running the setup.sh script, but I was switching branches and was assuming that I needed to rerun the setup.sh script. This gave the corresponding errors. However, I am still able to run the planners fine and there has not been any warning/errors when running the global planner launch file. I am going to assume that it is harmless for now and so I closed the issue I opened in #297. I can reopen the issue if we want to discuss the details further there.

yanhaoy, 2022-06-16:
If a later run fails, it won't overwrite the previously installed library, so you can still run it. As for your structure, you should have all these files like hsl* or ma* under quad-sdk/external/ipopt/coinhsl/ . Please let me know if this solves the problem, thanks!

AndrewZheng-1011, 2022-06-16:
I edited my previous post as I had lay out my folder structure wrong. In regard to the error, I realized I had a coinhsl directory within my original coinhsl directory. After removing it, the error fixed itself. Thanks for the help! Additionally, would you like me to close the issue page or keep the issue up until there exists a cleaner folder structure for loading the model?

yanhaoy, 2022-06-17:
Glad to hear you solved the problem. Yes, as @jrenaf said, the loading different robots bug will be fixed in the next merge, I'll close the issue. Thanks!

closed 2022-06-17 by yanhaoy

prakharg01, 2022-07-06:
Hello!

I have added a new robot recently to this SDK, I am able to load the robot in simulation after all the changes made!
But the initial position of the robot is not sitting, it falls from a standing pose. Due to this the robot state initially is very unstable and I can't run the stand mode.

Where in the SDK do I have to add the initial config of the robot?

nanbwrn, 2023-10-19:
Have you resolved the issue where Gazebo cannot display the robot model when gui:=true is set? @lnotspotl

nanbwrn, 2023-10-19:
I still can't display the model in Gazebo following the method mentioned in the issue.