robomechanics/quad-sdk #300: New Robot not loading with correct initial pos

prakharg01, 2022-07-06:
Hello!

I have added a new robot recently to this SDK, I am able to load the robot in simulation after all the changes made!
But the initial position of the robot is not sitting, it falls from a standing pose. Due to this the robot state initially is very unstable and I can't run the stand mode.

Where in the SDK do I have to add the initial config of the robot?

jrenaf, 2022-07-08:
You may refer to rosparam /robot_driver/sit_joint_angles in quad_utils/config/spirit.yaml. If you add a new robot type, you should copy the spirit.yaml to a new yaml file of your new robot type and change the parameters accordingly .

prakharg01, 2022-07-08:
Hey Thanks!

Yeah, I have done that, but changing "sit_joint_angles" doesn't change anything in simulation. I see the robot always landing in the same standing pose initially and then falling.

I see the urdf files of both a1 and spirit have "origin rpy" changed from standard value 0 to some values 1.57 radians or 90 degrees. Is that hardcoded in the urdf, or changing sit_joint_angles should take care of that?

Thanks,
Prakhar

jrenaf, 2022-07-10:
You are right, it is hardcoded in the urdf. Origin rpy and axis of some joints are modified to ensure that all legs are leveled at the inital pose.

closed 2022-07-15 by jrenaf