robomechanics/quad-sdk #303: mocap_optitrack problem

gearguan, 2022-07-25:
Hi there,
I am trying to deploy the quad-sdk project on hardware, but I don't have OptiTrack mocap hardware, is this necessary?
It seems that there is no odometer in quad-sdk, so is the configuration of the robot in the world coordinate system only available in this way?
Looking forward to the reply, thanks a lot! :)

ardalantj, 2022-07-26:
Hi @gearguan , 
Thank you for using quad-sdk. We have a couple research projects focused on on-board state estimation, but they are not quite ready for hardware deployment yet. If your robot has an on-board tracking camera you could use that for state estimation. This should work indoor if you have enough features in the environment. Otherwise, you could write your own state estimator using IMU and leg odometry. We have a template for adding state estimators inside the robot driver package. I hope this helps. Let us know if you have any other questions.

ardalantj, 2022-08-18:
Closing this issue. Please open a new issue if you have any follow up questions.

closed 2022-08-18 by ardalantj