robomechanics/quad-sdk #306: Use rosdep for ros package dependencies

qw8z5fUB, 2022-08-01:
Hi there, thanks for open sourcing these packages!

It looks like the following apt installs can all be handled by rosdep, by adding the appropriate rosdep keys to the appropriate package.xml files:

```
quad-sdk$ grep -r apt | grep ros-melodic
external/setup_deps.sh:sudo apt install -y ros-melodic-urdf
external/setup_deps.sh:sudo apt install -y ros-melodic-teleop-twist-joy
local_planner/setup_deps.sh:sudo apt install -y ros-melodic-grid-map
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-controller-manager -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-joint-state-controller -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-gazebo-ros-pkgs -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-ros-control -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-gazebo-ros-control -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-effort-controllers -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-robot-state-publisher -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-imu-tools -y
quad_simulator/setup_deps.sh:sudo apt install -y ros-melodic-message-to-tf -y
global_body_planner/setup_deps.sh:sudo apt install -y ros-melodic-tf
global_body_planner/setup_deps.sh:sudo apt install -y ros-melodic-tf2
global_body_planner/setup_deps.sh:sudo apt install -y ros-melodic-tf2-geometry-msgs
global_body_planner/setup_deps.sh:sudo apt install -y ros-melodic-grid-map
quad_utils/setup_deps.sh:sudo apt install -y ros-melodic-plotjuggler-ros
```
As a user I'm happy to `rosdep install --from-paths src`, however I'm more reluctant to run a bunch of setup_deps.sh scripts, many of which are using sudo privileges. Moving the above apt installs to be handled by rosdep would be one step towards minimising the use of these setup_deps.sh files.

ardalantj, 2022-08-02:
Hi, thanks for using quad-sdk and your input. You are right and we plan on moving towards rosdep very soon.

ardalantj, 2022-08-05:
Closing this issue. Open another one if there are any follow ups.

closed 2022-08-05 by ardalantj