robomechanics/quad-sdk #307: High level control without local planner

itaouil, 2022-08-02:
Hi,

Is it possible to command the robot with velocities without the need of having the local planner running on the side?

jrenaf, 2022-08-03:
Hi, the local planner is an interface between the nonlinear mpc solver and the lower-level motor controller. It also contains a footstep planner block which figures out the foothold locations so it's indispensable either using higher-level global planner or the velocity command.

jcnorby, 2022-08-04:
To be clear, the local planner is exactly what translates something like velocity commands into executable information. So if what you really want is high level control via velocity commands, the local planner is specifically designed to make that sort of API possible (see the examples of controlling the robot with a joystick).

closed 2022-08-04 by jcnorby

itaouil, 2022-08-05:
Hi,

Thanks for the replies :).

One last question. Can the the controller work blindly then? Or does it always require a local planner that has access to some sort of height map to select a reference foothold?