robomechanics/quad-sdk #309: Could not find qpOASES

jeanbuntu, 2022-08-07:
```
CMake Error at quad-sdk/local_planner/CMakeLists.txt:34 (find_library):
  Could not find QPOASES using the following names: qpOASES

```
What is it and what do I do for it? 
Completed step 1 through 6 on the wiki, with the exception of using catkin_make
Seems to throw lesser errors.

I've tried catkin build also, catkin build errors below
```
...............................................................................
Failed     << quad_utils:make                      [ Exited with code 2 ]      
Failed    <<< quad_utils                           [ 15.8 seconds ]            
Abandoned <<< gazebo_scripts                       [ Unrelated job failed ]    
Abandoned <<< global_body_planner                  [ Unrelated job failed ]    
Abandoned <<< nmpc_controller                      [ Unrelated job failed ]    
Abandoned <<< local_planner                        [ Unrelated job failed ]    
Abandoned <<< robot_driver                         [ Unrelated job failed ]    
[build] Summary: 8 of 14 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  9 packages succeeded with warnings.                       
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 36.4 seconds total.      
```

My linear solver setting was already ma27. Didn't need to change it in the code.

jrenaf, 2022-08-09:
Hi @jeanbuntu, Quad-sdk actually doesn't depend on qpOASES any more. If you have any problem with that dependency, a quick solution is to delete anything related to qpOASES in local planner CMakelist.

ardalantj, 2022-09-23:
@jeanbuntu Closing this issue due to inactivity.

closed 2022-09-23 by ardalantj