robomechanics/quad-sdk #311: Running Spirit 40

YuhanDong0806, 2022-08-12:
While I'm following the RML Guide: Running Spirit and run "source launch_robot_env.sh", there is no file called "launch_robot_env.sh", only "env.sh". When I source env.sh, it said that "calling env.sh without argument is not supported anymore. Instead spawn a subshell and source a setup file manually". What should I do to set up the environment now?

jrenaf, 2022-08-12:
Hi @YuhanDong0806, we actually didn't upload this file to the repo because it's mostly for RML internal use. If you want to know how to run Spirit 40 using quad-sdk, you can email me.

YuhanDong0806, 2022-08-13:
I see. Thank you for your quick response. I do want to know how to run Spirit 40 using quad-sdk. What's your email? Mine is yuhand@umcih.edu

jrenaf, 2022-08-15:
Email sent to you.

ardalantj, 2022-09-23:
Closing this issue due to inactivity.