robomechanics/quad-sdk #312: CasADi's code-generation from matlab to C++

WDouven, 2022-08-15:
hello, I can achieve the CasADi's code-generation from matlab to C++, but how to ultilize the code generated in my project as a part of it ? Thanks in advance!

jrenaf, 2022-08-15:
Hi, you can put Casadi generated files under [here](https://github.com/robomechanics/quad-sdk/tree/main/nmpc_controller/src/gen) and change what you want it to be loaded in `quad_nlp_utils.cpp` accordingly.

WDouven, 2022-08-16:
thanks for your response, I will try it. Can it be realized that solve the NLP cost function without ipopt but with the generated code only?

jrenaf, 2022-08-17:
You may take a reference at #304, but in general there are always a bunch of optimization solvers available to fit into our setup.

WDouven, 2022-08-18:
Thanks, I will keep trying.

closed 2022-08-18 by WDouven