robomechanics/quad-sdk #315: NMPC solving fail

samjupo, 2022-08-28:
I follow the guidence [https://github.com/robomechanics/quad-sdk/wiki/1.-Getting-Started-with-Quad-SDK](url) and the code was compiled without errors. 
But I meet the warning **NMPC solving fail** as the images show when I use the following the commands:

- catkin_make run_tests
-  roslaunch quad_utils quad_plan.launch reference:=twist logging:=true

image-1: quadruped stands
[image]

image-2: output: **NMPC solving fail** with roslaunch quad_utils quad_plan.launch reference:=twist logging:=true
[image]

image-3: output: catkin_make run_tests
[image]

jcnorby, 2022-08-28:
Looks like this is likely a duplicate of #288, can you try that solution and see if it solves your problem?

ardalantj, 2022-09-23:
@samjupo Any updates on this?

ardalantj, 2022-10-10:
closing due to inactivity

closed 2022-10-10 by ardalantj