robomechanics/quad-sdk #316: Loading Custom Terrain Map Files

AndrewZheng-1011, 2022-08-29:
Hey,

I'm trying to load different worlds into this framework and following the guidelines through your wiki. I am using Solidworks as my CAD software to create the three files (.sldprt, .stl, .ply). The issue I am having is that the rviz perceived ground and the solidworks stl mesh are not the same as shown below:

[image]
[image]

I believe this may be an issue of the origin of the .ply file not being the same as the .stl file. For .stl file I enabled the option "Do not translate STL output data to positive space". However, I could not find similar options for that when saving the file as a .ply file in solidworks. Could I get some suggestion on how you guys created your custom terrains and why there is this mismatch of terrains in rviz vs. gazebo?

jrenaf, 2022-08-30:
Hi, Gazebo and Rviz terrains are loaded from different sources in quad-sdk and when exporting .ply and .stl from Solidworks might have different origins issue. One quick solution to try is just fixing the origin of Gazebo terrain from World file.

to-the-abyss, 2022-08-30:
As a side note, a quick fix that worked for me was to set your terrain's height to be the same as that of the original terrain's (0.1m)

AndrewZheng-1011, 2022-08-30:
Thanks for the answer. I solved the solution by setting origin of world file, in this case, to (0,0,0) for xyz when originally at (0,0,-0.1). A small mistake on my part that I somehow overlooked when I made changes to exporting the stl file from the origin.

Results look good
[image]
[image]

closed 2022-08-30 by AndrewZheng-1011