robomechanics/quad-sdk #318: [robot_1/local_planner-1] process has died

XIAO6454, 2022-09-06:
I have successfully launch the rviz environment with "roslaunch quad_utils quad_gazebo.launch", and the robot was able to stand up with "rostopic pub /robot_1/control/mode std_msgs/UInt8 "data: 1".
However I got the following error while running the command "roslaunch quad_utils quad_plan.launch reference:=twist logging:=true"

[robot_1/local_planner-1] process has died [pid 24202, exit code -11, cmd /home/ao/quad_ws/devel/lib/local_planner/local_planner_node __name:=local_planner __log:=/home/ao/.ros/log/848cb238-2dbf-11ed-9ada-2c16dba83582/robot_1-local_planner-1.log].
log file: /home/ao/.ros/log/848cb238-2dbf-11ed-9ada-2c16dba83582/robot_1-local_planner-1*.log

Any idea?

jrenaf, 2022-09-06:
Hi @XIAO6454, it is difficult to directly debug from this error. Any other errors showed up when you run the command?

XIAO6454, 2022-09-07:
> Hi @XIAO6454, it is difficult to directly debug from this error. Any other errors showed up when you run the command?

Thank you for your reply! It seems no other errors occurred. The following is what the terminal printed when I run this command, you can see that the error showed up at the end where the process is stuck:

(base) ao@ao:~/quad_ws$ roslaunch quad_utils quad_plan.launch reference:=twist logging:=true
... logging to /home/ao/.ros/log/5ec284b6-2e45-11ed-ab23-e0d464396f1b/roslaunch-ao-10427.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ao:39501/

SUMMARY
========

PARAMETERS
 * /robot_1/enable_statistics: True
 * /robot_1/global_body_planner/backup_ratio: 0.5
 * /robot_1/global_body_planner/dt: 0.03
 * /robot_1/global_body_planner/dz0_max: 2.0
 * /robot_1/global_body_planner/dz0_min: 1.0
 * /robot_1/global_body_planner/g: 9.81
 * /robot_1/global_body_planner/goal_state: [5.0, 0.0]
 * /robot_1/global_body_planner/grf_max: 5.0
 * /robot_1/global_body_planner/grf_min: 4.0
 * /robot_1/global_body_planner/h_max: 0.375
 * /robot_1/global_body_planner/h_min: 0.075
 * /robot_1/global_body_planner/h_nom: 0.3
 * /robot_1/global_body_planner/mass: 13.0
 * /robot_1/global_body_planner/max_planning_time: 0.2
 * /robot_1/global_body_planner/mu: 0.25
 * /robot_1/global_body_planner/num_calls: 1
 * /robot_1/global_body_planner/num_leap_samples: 10
 * /robot_1/global_body_planner/pos_error_threshold: 25
 * /robot_1/global_body_planner/replanning: True
 * /robot_1/global_body_planner/robot_h: 0.05
 * /robot_1/global_body_planner/robot_l: 0.3
 * /robot_1/global_body_planner/robot_w: 0.3
 * /robot_1/global_body_planner/startup_delay: 2.0
 * /robot_1/global_body_planner/t_s_max: 0.25
 * /robot_1/global_body_planner/t_s_min: 0.12
 * /robot_1/global_body_planner/trapped_buffer_factor: 7
 * /robot_1/global_body_planner/traversability_threshold: 0.3
 * /robot_1/global_body_planner/update_rate: 20
 * /robot_1/global_body_planner/v_max: 2.0
 * /robot_1/global_body_planner/v_nom: 0.75
 * /robot_1/grid_map_visualization/grid_map_topic: /terrain_map
 * /robot_1/grid_map_visualization/grid_map_visualizations: [{'type': 'point_...
 * /robot_1/local_footstep_planner/duty_cycles: [0.5, 0.5, 0.5, 0.5]
 * /robot_1/local_footstep_planner/foothold_obj_threshold: 0.6
 * /robot_1/local_footstep_planner/foothold_search_radius: 0.25
 * /robot_1/local_footstep_planner/grf_weight: 0.5
 * /robot_1/local_footstep_planner/ground_clearance: 0.07
 * /robot_1/local_footstep_planner/hip_clearance: 0.1
 * /robot_1/local_footstep_planner/obj_fun_layer: traversability
 * /robot_1/local_footstep_planner/period: 0.36
 * /robot_1/local_footstep_planner/phase_offsets: [0.0, 0.5, 0.5, 0.0]
 * /robot_1/local_footstep_planner/standing_error_threshold: 0.03
 * /robot_1/local_planner/cmd_vel_filter_const: 0.1
 * /robot_1/local_planner/cmd_vel_scale: 1.0
 * /robot_1/local_planner/desired_height: 0.27
 * /robot_1/local_planner/horizon_length: 26
 * /robot_1/local_planner/last_cmd_vel_msg_time_max: 2.0
 * /robot_1/local_planner/stand_cmd_vel_threshold: 0.1
 * /robot_1/local_planner/stand_pos_error_threshold: 0.05
 * /robot_1/local_planner/stand_vel_threshold: 0.1
 * /robot_1/local_planner/timestep: 0.03
 * /robot_1/local_planner/toe_radius: 0.02
 * /robot_1/local_planner/update_rate: 333
 * /robot_1/local_planner/use_twist_input: True
[... 177 lines cut ...]
process[robot_1/local_planner-1]: started with pid [10445]
process[robot_1/rosbag_record-2]: started with pid [10446]
process[robot_1/rosbag_record_archive-3]: started with pid [10447]
process[robot_1/plan_publisher-4]: started with pid [10448]
complexity_schedule = 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Setting: "make_parameter_nodual" is not a valid setting for Option: fixed_variable_treatment. Check the option documentation.

### fixed_variable_treatment (String) ###
Category: NLP
Description: Determines how fixed variables should be handled.
Valid Settings:
	make_parameter (Remove fixed variable from optimization variables)
	make_constraint (Add equality constraints fixing variables)
	relax_bounds (Relax fixing bound constraints)
Default: "make_parameter"
Tried to set Option: max_wall_time. It is not a valid option. Please check the list of available options.
_[robot_1/local_planner-1] process has died [pid 10445, exit code -11, cmd /home/ao/quad_ws/devel/lib/local_planner/local_planner_node __name:=local_planner __log:=/home/ao/.ros/log/5ec284b6-2e45-11ed-ab23-e0d464396f1b/robot_1-local_planner-1.log].
log file: /home/ao/.ros/log/5ec284b6-2e45-11ed-ab23-e0d464396f1b/robot_1-local_planner-1*.log_
^C[robot_1/plan_publisher-4] killing on exit
[robot_1/rosbag_record_archive-3] killing on exit
[robot_1/rosbag_record-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
(base) ao@ao:~/quad_ws$

P1terQ, 2022-09-09:
I've the same problem. Before this i met with _nmpc solving fail_. i reinstall the related packages, somehow it disappeared and come up with this one

ardalantj, 2022-09-16:
Has this been resolved?

XIAO6454, 2022-09-19:
> Has this been resolved?
It disappeared but _nmpc solving fail_ was showed up.

ardalantj, 2022-09-23:
@XIAO6454 Any updates on this?

ardalantj, 2022-10-10:
Closing due to inactivity

closed 2022-10-10 by ardalantj