robomechanics/quad-sdk #319: Robot fall across the ground

P1terQ, 2022-09-08:
`catkin build` is OK. However,after i  `roslaunch quad_utils quad_gazebo.launch`, the robot directly fall across the ground in rviz as if there is no collision constraints.And there is no warning or something. Anyone know what's wrong?

P1terQ, 2022-09-08:
After i set gui = true, the robot still falls. And i'm getting warnings:

> [Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
> [Wrn] [ModelDatabase.cc:212] Unable to connect to model database using [http://models.gazebosim.org//database.config]. Only locally installed models will be available.
> [Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [model:///meshes/Spirit40_Lower_Leg_transformed.stl]
> [Wrn] [SystemPaths.cc:464] File or path does not exist [""] [model:///meshes/Spirit40_Lower_Leg_transformed.stl]
> [Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
> [Wrn] [ModelDatabase.cc:212] Unable to connect to model database using [http://models.gazebosim.org//database.config]. Only locally installed models will be available.
> [Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [model:///meshes/Spirit40_Upper_leg_transformed.stl]
> [Wrn] [SystemPaths.cc:464] File or path does not exist [""] [model:///meshes/Spirit40_Upper_leg_transformed.stl]
> [Err] [Visual.cc:3046] No mesh specified

P1terQ, 2022-09-09:
think i've solved the problem. The key is to add the following lines to the environment variables
`export GAZEBO_MODEL_PATH=:/home/peter/catkin_ws/src/quad-sdk/quad_simulator/spirit_description:/home/peter/catkin_ws/src/quad-sdk/quad_simulator/other/sensor_description:/home/peter/catkin_ws/src/quad-sdk/quad_simulator/other/objects_description:/home/peter/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds`

closed 2022-09-09 by P1terQ

goldbattle, 2023-01-15:
This seems to happen if you move the folder and don't re-run the setup script, or reset your bashrc. This model path is added in the simulator setup_deps.sh file, thus make sure to `source ~/.bashrc` after running the setup script too
https://github.com/robomechanics/quad-sdk/blob/b11ce97744a55a0116aebd04bd1c4b50e39849a5/quad_simulator/setup_deps.sh#L1-L9