robomechanics/quad-sdk #320: Linking issue with

dkanou, 2022-09-16:
I am getting an annoying linking error:

quad-sdk/nmpc_controller/include/nmpc_controller/quad_nlp.h:15:10: fatal error: IpIpoptData.hpp: No such file or directory

And I am not sure why as all installations and setup bash scripts ran as instructed.

dkanou, 2022-09-17:
HI all, actually it seems that IpIpoptData.hpp does not exist in the IPOPT HSL solver.  So it might not be a linking issue but rather a change on the IPOPT package?

eorhon, 2022-09-18:
Yeah, I get the same error and cant find a way to get rid of it. I did manage the build 2 months ago. However, when i try to reinstall the quad-sdk, i repeatedly get this error :(

ardalantj, 2022-09-19:
This may be due to the update on the IPOPT side as we have not made any changes on the main related to this. We are investigating this and will post a solution soon.

ardalantj, 2022-09-20:
The issue appear to be with the latest update of IPOPT. Simply changing line 19 of /external/setup_deps.sh to "./coinbrew fetch Ipopt@3.13 --no-prompt" to use release 3.13 should work. We will push this change along with some updates to dependency installations soon.

fegaeze, 2022-09-21:
Hello! Tried your suggestion @ardalantj. Still having issues with the build. Did this work for someone else?! This is a new installation for me.

Boogard2021, 2022-09-21:
Hi. I also tried @ardalantj's suggestion and it did not work. I still have this error.

dkanou, 2022-09-22:
Hi all, OK it compiles if you change also line 29 accordingly, i.e., Ipopt@3.13

ardalantj, 2022-09-22:
@dkanou Great ! thanks for sharing that.

ardalantj, 2022-09-22:
@Boogard2021 @fegaeze Let me know if this works for you so we can close the issue.

fegaeze, 2022-09-22:
Hello! Just tried to change the version for both lines. Everything seemed fine during setup. When I tried to catkin build, nmpc_controller failed. Maybe there are some other issues? Still the same error about Missing IpIpoptData.hpp

eorhon, 2022-09-22:
I build ipopt by hand. By opening terminal at external folder and write `./configure` and then `make` and then `sudo make install` after that i recieve error in #318. To solve it I reinstall some dependencies and write `sudo apt-get update` and then `sudo apt-get upgrade` which solve my issues. However, I am a newbie at Ubuntu, I do not know what exactly solved my issues. Hope this helps somehow.

fegaeze, 2022-09-22:
@ardalantj Tried a clean install. Still didn't work. Then tried fix by @eorhon, and it worked.

Boogard2021, 2022-09-22:
@eorhon Thanks for your suggestion. It compiled successfully.

closed 2022-09-23 by ardalantj

ChangerC77, 2022-11-18:
> @eorhon, @dkanou, the issue appear to be with the latest update of IPOPT. Simply changing line 19 of /external/setup_deps.sh to "./coinbrew fetch Ipopt@3.13 --no-prompt" to use release 3.13 should work. We will push this change along with some updates to dependency installations soon.

after done, more errors occured:
Errors     << nmpc_controller:make /home/ros/cmu/quad_sdk/logs/nmpc_controller/build.make.014.log                                                                                                          
In file included from /home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/src/quad_nlp_utils.cpp:1:0:
/home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/include/nmpc_controller/quad_nlp.h:15:10: fatal error: IpIpoptData.hpp: No such file or directory
 #include <IpIpoptData.hpp>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nmpc_controller.dir/src/quad_nlp_utils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/src/quad_nlp.cpp:1:0:
/home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/include/nmpc_controller/quad_nlp.h:15:10: fatal error: IpIpoptData.hpp: No such file or directory
 #include <IpIpoptData.hpp>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nmpc_controller.dir/src/quad_nlp.cpp.o] Error 1
In file included from /home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/src/nmpc_controller.cpp:1:0:
/home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/include/nmpc_controller/nmpc_controller.h:16:10: fatal error: IpIpoptApplication.hpp: No such file or directory
 #include "IpIpoptApplication.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nmpc_controller.dir/src/nmpc_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/nmpc_controller.dir/all] Error 2
make: *** [all] Error 2
cd /home/ros/cmu/quad_sdk/build/nmpc_controller; catkin build --get-env nmpc_controller | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

...........................................................................................................................................................................................................
Failed     << nmpc_controller:make               [ Exited with code 2 ]                                                                                                                                    
Failed    <<< nmpc_controller                    [ 3.2 seconds ]                                                                                                                                           
Abandoned <<< local_planner                      [ Unrelated job failed ]                                                                                                                                  
[build] Summary: 11 of 13 packages succeeded.                                                                                                                                                              
[build]   Ignored:   None.                                                                                                                                                                                 
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 5.6 seconds total.