robomechanics/quad-sdk #327: Incorrect joint limits in A1 urdf

qw8z5fUB, 2022-10-31:
Hi there, I've noticed the joint limits specified in a1.urdf do not seem to match those of the actual robot.

Hence I am unsure if results I achieve in gazebo accurately represent what is possible in the real world.

For instance, from [a1.urdf](https://github.com/robomechanics/quad-sdk/blob/main/quad_simulator/a1_description/urdf/a1.urdf#L641-L648)
```
  <joint name="2" type="revolute">
    <origin rpy="0 1.57079632679 0" xyz="0 0.0838 0"/>
    <parent link="hip1"/>
    <child link="upper1"/>
    <axis xyz="0 -1 0"/>
    <dynamics damping="0" friction="0"/>
    <limit effort="33.5" lower="-6.28318530718" upper="6.28318530718" velocity="21"/>
  </joint>
```
In this urdf the joint can rotate from `-2*pi` to `+2*pi`, however from the [unitree urdf](https://github.com/unitreerobotics/unitree_ros/blob/master/robots/a1_description/urdf/a1.urdf) I believe this joint only has a 300 degree range of motion in reality. Hence quad-sdk may be providing motion commands which are physically impossible for the robot to follow.

jrenaf, 2022-10-31:
Hi @qw8z5fUB, I think answers to Issue #325 will be helpful to you. If you have further questions, you may ask there.

qw8z5fUB, 2022-10-31:
Hello, thanks, yes I did see that issue, however that relates more to the conventions used to create the a1 urdf (i.e. ghost conventions), while this issue relates to the joint limits in a1.urdf being incorrect (regardless of conventions used to specify them).

I thought this was worth raising separately, so that other users are aware that results from gazebo may not be replicable in the real world, given some of the kinematic constraints of the real robot have been ignored. 

FYI @thvhauwe.

jcnorby, 2022-11-05:
Thanks for the feedback! I agree, even if the conventions are different the capabilities should match the actual system (since that's just a matter of updating some coefficients). We'll see if this can make it into the next PR into devel/main, although as always you are welcome to submit a PR if you have those numbers yourself.

With respect to replicating results on hardware - the velocity and torque limits on Spirit and the A1 are pretty similar, and I've noticed that the lack of hard joint limits on Spirit actually make the system _less_ stable rather than moreso (particularly when leaping, as it can be easy for the legs to hyper-extend on landing). So hopefully the current setup errs on the side of being too conservative rather than too optimistic.

qw8z5fUB, 2022-11-09:
Hello, just to update. I've had a go at updating the limits, and I think I have the correct values now. For the hip->upper joints I subtracted pi/2 from the unitree limits, and for the upper->lower joints I added pi.
I tested using the joint publisher gui to confirm the leg ranges of motion visually match those from the unitree a1.urdf.
I don't think I'll be able to put a pull request together, but I can provide a diff (below). I've updated the sdf and urdf - I imagine the xacro will also need to be updated.
```
diff --git a/quad_simulator/a1_description/sdf_mesh/a1.sdf b/quad_simulator/a1_description/sdf_mesh/a1.sdf
index de06e4c1..34dacbdc 100644
--- a/quad_simulator/a1_description/sdf_mesh/a1.sdf
+++ b/quad_simulator/a1_description/sdf_mesh/a1.sdf
@@ -248,8 +248,8 @@
       <axis>
         <xyz>0 -1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>-2.6179938779914944</lower>
+          <upper>2.6179938779914944</upper>
           <effort>33.5</effort>
           <velocity>21</velocity>
         </limit>
@@ -320,8 +320,8 @@
       <axis>
         <xyz>0 1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>0.445058959258554</lower>
+          <upper>2.2252947962927703</upper>
           <effort>33.5</effort>
           <velocity>21</velocity>
         </limit>
@@ -559,8 +559,8 @@
       <axis>
         <xyz>0 -1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>-2.6179938779914944</lower>
+          <upper>2.6179938779914944</upper>
           <effort>33.5</effort>
           <velocity>21</velocity>
         </limit>
@@ -631,8 +631,8 @@
       <axis>
         <xyz>0 1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>0.445058959258554</lower>
+          <upper>2.2252947962927703</upper>
           <effort>33.5</effort>
           <velocity>21</velocity>
         </limit>
@@ -870,8 +870,8 @@
       <axis>
         <xyz>0 -1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>-2.6179938779914944</lower>
+          <upper>2.6179938779914944</upper>
           <effort>33.5</effort>
           <velocity>21</velocity>
         </limit>
@@ -942,8 +942,8 @@
       <axis>
         <xyz>0 1 0</xyz>
         <limit>
-          <lower>-6.28319</lower>
-          <upper>6.28319</upper>
+          <lower>0.445058959258554</lower>
[... 78 lines cut ...]
     <dynamics damping="0" friction="0"/>
-    <limit effort="33.5" lower="-6.28318530718" upper="6.28318530718" velocity="21"/>
+    <limit effort="33.5" lower="0.445058959258554" upper="2.2252947962927703" velocity="21"/>
   </joint>
   <link name="toe2">
     <visual name="toe2_visual">
@@ -1004,7 +1004,7 @@
     <child link="upper3"/>
     <axis xyz="0 -1 0"/>
     <dynamics damping="0" friction="0"/>
-    <limit effort="33.5" lower="-6.28318530718" upper="6.28318530718" velocity="21"/>
+    <limit effort="33.5" lower="-2.6179938779914944" upper="2.6179938779914944" velocity="21"/>
   </joint>
   <link name="lower3">
     <visual name="lower3_visual">
@@ -1032,7 +1032,7 @@
     <child link="lower3"/>
     <axis xyz="0 1 0"/>
     <dynamics damping="0" friction="0"/>
-    <limit effort="33.5" lower="-6.28318530718" upper="6.28318530718" velocity="21"/>
+    <limit effort="33.5" lower="0.445058959258554" upper="2.2252947962927703" velocity="21"/>
   </joint>
   <link name="toe3">
     <visual name="toe3_visual">
```

ardalantj, 2023-07-17:
Closing due to inactivity.

qw8z5fUB, 2023-07-18:
My understanding is that this is still an existing error in the codebase, I suppose it will not be addressed.

ologandavid, 2023-07-19:
Hi @qw8z5fUB, 
Did you get working joint limits that you could share with us so we could update the code base with them or are you still expereiencing trouble?

Best, 
David Ologan

qw8z5fUB, 2023-07-20:
Hello, I didn't do any further work beyond the diff I posted and associated visual check. I don't plan to look further into it, perhaps the appropriate action is to close as not planned (won't fix). At least the issue is documented in this ticket, and if anyone hits this in future they can reopen and look at a PR.

ologandavid, 2023-07-20:
Sounds good. I can look into fixing it on my end as well. 

Best, 
David

closed 2023-07-27 by ologandavid