robomechanics/quad-sdk #331: errors when catkin build

ChangerC77, 2022-11-18:
Download and compile Ipopt and HSL, and then compile quad_sdk, the following problems occured:
Errors     << nmpc_controller:make /home/ros/cmu/quad_sdk/logs/nmpc_controller/build.make.010.log                                                                                                          
In file included from /home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/src/quad_nlp_utils.cpp:1:0:
/home/ros/cmu/quad_sdk/src/quad-sdk/nmpc_controller/include/nmpc_controller/quad_nlp.h:15:10: fatal error: IpIpoptData.hpp: No such file or directory
 #include <IpIpoptData.hpp>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nmpc_controller.dir/src/quad_nlp_utils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/nmpc_controller.dir/all] Error 2
make: *** [all] Error 2
cd /home/ros/cmu/quad_sdk/build/nmpc_controller; catkin build --get-env nmpc_controller | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

...........................................................................................................................................................................................................
Failed     << nmpc_controller:make                 [ Exited with code 2 ]                                                                                                                                  
Failed    <<< nmpc_controller                      [ 2 minutes and 0.2 seconds ]                                                                                                                           
Abandoned <<< local_planner                        [ Unrelated job failed ]                                                                                                                                
[build] Summary: 12 of 14 packages succeeded.                                                                                                                                                              
[build]   Ignored:   None.                                                                                                                                                                                 
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 3 minutes and 27.7 seconds total.

ChangerC77, 2022-11-18:
There are too many errors and third-party packages, so it is strongly recommended to use Docker

andrewhstovell, 2022-11-22:
+1
Getting the exact same issue. I have tried steps in #320 

**Everything i've done:**

Downloaded, extracted and HSL (Personal Licence)
installed with: 
```
cd /PATH/TO/coinhsl-archive-YYYY.MM.DD
./configure
make
sudo make install
```

then:
`cp -r /usr/local/libs/libcoinhsl.so ~/catkin_ws/src/quad-sdk/external/ipopt/coinhsl`

ran `chmod +x setup.sh && ./setup.sh` without issue

Modified quad-sdk/external/setup_deps.sh as follows:
...
_19_ `./coinbrew fetch Ipopt@3.13 --no-prompt`
...
_29_ `./coinbrew build Ipopt@3.13 --tests none --prefix=/usr/local --no-prompt --parallel-jobs=8`
...

have since tried `sudo apt update` + `sudo apt upgrade`
as well as deleting devel and build dir's and building from scratch.

Could this be a HSL licence/code issue?

andrewhstovell, 2022-11-22:
I'm not sure where i got the idea to copy + rename the .so to the external/ipopt. I realised that i was supposed to copy + rename the downloaded (and extracted) coinhsl-archive-YYYY.MM.DD instead. 

_Doing this,_ 
changing the lines (19 + 29) in setup_deps, 
running `sudo chmod +x setup.sh && ./setup.sh` in quad-sdk
`rm -r build devel` from catkin_ws to start fresh
and running `catkin build` 

has resulted in a successful build for me.

jcnorby, 2023-01-28:
Glad that something worked! And agreed that providing a docker image would be useful for debugging issues like this. Most of the IPOPT difficulties come from the setup script pulling from the IPOPT repo, which updates from time to time. I believe it is an active todo to handle this in a more scalable way.

That said, some of these changes were updated on the devel branch shortly after this issue was logged - is anyone able to verify that those changes solve this issue? If no one is actively having seeing this then we can probably close the issue, but we can leave it open for a bit in case others are running into the same problem.

jcnorby, 2023-01-28:
I also updated the documentation to be a bit more clear what the IPOPT paths are supposed to be prior to running the setup script, that seems to have been a source of confusion for many. Yet another reason to provide docker support!

closed 2023-02-06 by jcnorby