robomechanics/quad-sdk #332: A question about adding a new robot

HuShanzhi, 2022-11-29:
I tried to add the `solo12` robot to the toolkit. No error was prompted during runtime, but the robot could not display normally like this: (The robot model flashes and disappears)

[image]

I can modify the `fixed frame` property to make the robot and the map display separately.

[image]

[image]

Any help will be appreciated！

HuShanzhi, 2022-11-29:
**Describe the problem again:**

By changing `<gravity>1</gravity>` to `<gravity>0</gravity>` in the sdf file to cancel gravity, I get the following result:

[image]

Therefore, it is not the robot model that flashes, but the robot does not get support from the ground after falling due to gravity, and then falls until it disappears.

I am a rookie of ros, and I will be grateful for any help！

jcnorby, 2022-11-29:
You are correct that the issue with the ground - this is what happens when gazebo can't find the path to the terrain mesh file. If it can't find this file it doesn't add the ground into the simulation, thus the falling through the floor issue.

Check that your gazebo model path in your .bashrc is a valid path that points to the world .stl file - this should be added by the setup script but it seems like something isn't working. You can also launch the simulation with gui:=true to bring up the gazebo gui, this should verify if gazebo was able to find the terrain file or not.

HuShanzhi, 2022-11-30:
Thank you for your reply！I checked my .bashrc, the following contents are at the end of the file:
```shell
source /opt/ros/noetic/setup.bash
export GAZEBO_MODEL_PATH=:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/spirit_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/other/sensor_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/other/objects_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds
export GAZEBO_MODEL_PATH=:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/spirit_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/other/sensor_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/other/objects_description:/home/hu/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds:/home/hu/quad_noetic_ws/src/quad-sdk/quad_simulator/spirit_description:/home/hu/quad_noetic_ws/src/quad-sdk/quad_simulator/other/sensor_description:/home/hu/quad_noetic_ws/src/quad-sdk/quad_simulator/other/objects_description:/home/hu/quad_noetic_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds
```
For my computer space, path `/home/hu/quad_noetic_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds` is valid. 

Besides, I also attempt to launch the simulation with gui:=true, and get the following result:

[image]

Does the figure above indicate that gazebo can find the terrain file？

By the way, do I need to configure anything other than what is mentioned in the tutorial?  For example, configuring the relationship between robots and the world in `<robot_name>.sdf`.

HuShanzhi, 2022-11-30:
I seem to have found the reason. It may be caused by an error in the setting of the `<collision>` element.

For example, I changed the `<collision>` element of the `body link` from
```shell
<collision name='body_collision'>
  <pose>0 0 0 0 -0 0</pose>
  <geometry>
    <mesh>
      <scale>1 1 1</scale>
      <uri>model://solo12_description/meshes/solo_12_base.stl</uri>
    </mesh>
  </geometry>
</collision>
```
to
```shell
<collision name='body_collision'>
  <pose>0 0 0 0 -0 0</pose>
  <geometry>
    <box>
      <size>0.335 0.24 0.104</size>
    </box>
  </geometry>
</collision>
```

The robot's body will not fall all the time, although its legs will still cross the ground.

[image]

However, the size and shape of the `<box>` element is obviously different from the .stl file in `<mesh>` element.  I don't want to change or I don't have the ability to change the size of the `<box>`. Can I directly use .stl files for collision detection?

Thank you in advance！

HuShanzhi, 2022-11-30:
[image]

Maybe the picture above illustrates my meaning better. If I increase the thickness (height) of the box, which is much greater than the thickness of the robot body. For example, change the `<size>` element from 
```shell
<size>0.335 0.24 0.104</size>
```
to
```shell
<size>0.335 0.24 1</size>
```
This will cause the robot to be jacked up and unable to contact the ground, as shown in the figure, even the feet cannot touch the ground.

I don't know if this is a childish problem, but I still hope you can give me some help.

jcnorby, 2022-12-06:
Ah, I see - looks like the issue is that Gazebo isn't getting collision detection data from your URDF. The XML tag <visual> determines what you see, and <collision> determines what is actually used for collision detection, thus the discrepancy if you switched it to box.

From your snippet it looks like the collision data was originally defined in the solo12_description package, and from your .bashrc it looks like you haven't told Gazebo where that package lives. This would mean that Gazebo just wouldn't add any collision data, thus the falling through the floor (and why this is seemingly fixed by adding a box which doesn't need a path). This would also explain why things look okay in RViz (since ROS knows the path to that package) but not in the Gazebo gui (since it doesn't have that path).

Try adding some additional paths to your bashrc similar to the spirit_description paths which are already there but pointing to your solo12_description package. Also you can delete all the duplicate paths which ended up there, that's due to a slight bug in our setup script.

jcnorby, 2023-01-28:
Closing due to inactivity and it seems the root cause was found (missing gazebo plugin paths), please update if this isn't the case.

closed 2023-01-28 by jcnorby