robomechanics/quad-sdk #333: NMPC Solving Fail (Different from #288)

andrewhstovell, 2022-12-05:
Hi guys,

We're using the Academic version of HSL (with ma57) i've placed the folder in /external/ipopt/
(I do this before running the setup scripts)

I've ensured that **npmc_controller.cpp** lines 214 and 216 are   
```
app_->Options()->SetStringValue("linear_solver", "ma57");
app_->Options()->SetNumericValue("ma57_pre_alloc", 1.5);
```

In **setup_deps.sh** i am using the `--latest-release` of IPopt.

I had no errors or issues while running the setup scripts, and building. 
I also stand the quadruped before running the local planner.

[image]

ologandavid, 2022-12-14:
I'm currently running ma27, and the NMPC doesn't fail. I would suggest requesting version ma27 from iPOPT and rebuilding. If the error persists, please let me know.

jcnorby, 2023-01-28:
Was this issue solved? If not, try using the devel branch, some IPOPT stuff was resolved there. If that still doesn't resolve it, let us know. It looks like something in the paths isn't lining up right - we may need to update our documentation with a clear statement of what the path to the solver files should be (for reference, quad-sdk/external/ipopt/coinhsl/ma27/ma27d.f should be a valid path, or likewise for ma57 if that is the solver you are using).

jcnorby, 2023-02-26:
Closing this due to inactivity, also it seems likely that the root cause was incorrect HSL paths and clarity has been added to the installation instructions.

closed 2023-02-26 by jcnorby