robomechanics/quad-sdk #334: The new robot moves randomly after launching

HuShanzhi, 2022-12-06:
[image]

Hello, 

After some attempts, my robot can be loaded successfully without penetrating the ground. However, its initialization state is abnormal. The robot should have been sitting down and still (before I subscribed the standing command), but it was in chaos, as shown in the figure above.

In addition, as a rookie in ROS, what knowledge should I learn before using this SDK?

Any help will be appreciated！

jcnorby, 2022-12-06:
Looks like the zeros/directions may be different. Refer to this [link](https://github.com/robomechanics/quad-sdk/wiki/Tutorial:-Adding-a-New-Type-of-Robot-to-Quad-SDK) regarding URDF changes - you'll need to either modify the URDF so that the zeros and directions align with our convention (see the Spirit urdf for reference) or add mappings to various places in the repo.

jcnorby, 2023-01-28:
Closing for lack of activity, let us know if the problem persists.

closed 2023-01-28 by jcnorby