robomechanics/quad-sdk #335: Parallel-jobs in Coinbrew causes IPOPT/Mumps builds to fail

yanhaoy, 2022-12-14:
Some updates in the Mumps code seem to cause failures when building with parallel jobs. I see you switched to IPOPT@3.13 in the devel branch, but that still doesn't work for me. Removing the parallel-jobs parameter on line 29 of quad-sdk/external/setup_deps.sh made it work.

Tested on a fresh install of Ubuntu 18.04 with ROS Melodic with parameters Coinbrew@2.0, IPOPT@latest version. I believe it will also work if you use Coinbrew@1.0 and IPOPT@3.13. The updated setup_deps.sh is as follows:

```
wget https://raw.githubusercontent.com/coin-or/coinbrew/v2.0/coinbrew
chmod u+x coinbrew
./coinbrew fetch Ipopt --latest-release --no-prompt
cd ..
if [ -d "./coinhsl" ] 
then
    echo "HSL found." 
	cp -r ./coinhsl ./coinbrew/ThirdParty/HSL/coinhsl
else
    echo "Warning: HSL not found."
fi
cd coinbrew
./coinbrew build Ipopt --latest-release --tests none --prefix=/usr/local --no-prompt
```

jcnorby, 2023-09-16:
Note - after making some local changes (i.e. blowing away my entire catkin ws), I ran into a similar issue but for a different error. IPOPT was using ma27 but would exit each solve after one iteration with "EXIT: Stopping optimization at current point as requested by user". Seems somewhat related to https://github.com/coin-or/Ipopt/issues/661 but I was building in release. The above fix of updating to coinbrew v2.0 and latest IPOPT solved it. My recommendation would be to update from v1.0 -> v2.0 and 3.13 -> 3.14 (the current latest). Would like to avoid always using latest since they could make a new one that breaks things (although that didn't prevent this from breaking...).

elpimous, 2023-09-16:
Will you fix it under noetic_devel_ekf_clean ?

jcnorby, 2023-09-16:
I believe the current plan is to PR to devel/master along with docker support, but feel to apply these changes to your branch if you are seeing the same issue.

elpimous, 2023-09-16:
ok !
same issue, corrected and working now !
`# Setup and build for IPOPT
cd ipopt
sudo apt-get install -y gcc g++ gfortran git patch wget pkg-config liblapack-dev libmetis-dev
if [ -d "./coinbrew" ] 
then
	sudo rm -r ./coinbrew
fi
mkdir coinbrew
cd coinbrew
wget https://raw.githubusercontent.com/coin-or/coinbrew/v2.0/coinbrew
chmod u+x coinbrew
./coinbrew fetch Ipopt@3.14 --no-prompt
cd ..
if [ -d "./coinhsl" ] 
then
    echo "HSL found." 
	cp -r ./coinhsl ./coinbrew/ThirdParty/HSL/coinhsl
else
    echo "Warning: HSL not found."
fi
cd coinbrew
./coinbrew build Ipopt --tests none --prefix=/usr/local --no-prompt --parallel-jobs=8
cd ../..`
Thanks @jcnorby

ologandavid, 2023-10-30:
Incoporated into devel in recent pull request. Closing as completed.

closed 2023-10-30 by ologandavid