robomechanics/quad-sdk #337: ros noetic : Exception message: libhsl.so: cannot open shared object file: No such file or directory

elpimous, 2023-01-27:
```
error  : [ WARN] [1674854216.950660915, 53.878000000]: NMPC solving fail
Exception of type: DYNAMIC_LIBRARY_FAILURE in file "/home/ylo2/catkin_ws/src/quad-sdk/external/ipopt/coinbrew/Ipopt/src/Common/IpLibraryLoader.cpp" at line 67:
 Exception message: libhsl.so: cannot open shared object file: No such file or directory
```
Hello the team, using noetic branch of quad-sdk.
I don"t have any errors on `./setup.sh`, neither on `catkin build` terminal.
here here what i did : 

```
cd quad-sdk
cp -r /home/ylo2/Documents/ma27-1.0.0 ./external/ipopt/coinhsl
chmod +x setup.sh && ./setup.sh
cd ~/catkin_ws
catkin build
```

when running `roslaunch quad_utils quad_plan.launch reference:=twist logging:=true` I have this error !!  

I have : 
```
app_->Options()->SetStringValue("linear_solver", "ma27");
app_->Options()->SetNumericValue("ma57_pre_alloc", 1.5);
```

tested too  issues : 
`Parallel-jobs in Coinbrew causes IPOPT/Mumps builds to fail #335 `
`nmpc solving fail #288`

I'm a bit lost !! any idea ?

elpimous, 2023-01-27:
Hey, solved !!
i copied directly downloaded `ma27  dir ` to   `catkin_ws/src/quad-sdk/external/ipopt/coinhsl/`
(to help, this ma27 only contents 2 files : ma27d.f and ma27s.f)
and when ./setup.sh, I saw no error on 

```
##################################################
### Configuring Ipopt 3.14.10 
##################################################

##################################################
### Building Ipopt 3.14.10 
##################################################
```
Enjoy

closed 2023-01-28 by elpimous

jcnorby, 2023-01-28:
Glad that worked! Although it looks like there was an issue with your installation which led to the paths being incorrect in the first place. Specifically, the command:

`cp -r /home/ylo2/Documents/ma27-1.0.0 ./external/ipopt/coinhsl`

should be 

`cp -r /home/ylo2/Documents/coinhsl ./external/ipopt/coinhsl`

If done correctly, the following should be a valid path:

`<path-to-quad-sdk>/external/ipopt/coinhsl/ma27/ma27d.f`

Once that is the case, running the install script should copy this file to the correct location.

elpimous, 2023-01-28:
Correct. Thanks for your interest.