robomechanics/quad-sdk #340: Bad robot structure (only white body), on Ylo2 robot

elpimous, 2023-01-30:
Hi the team,
I have a good xacro model , (a modification of your A1_description)
working perfectly on rviz, and joints tested : 
[image]
[image]

elpimous, 2023-01-30:
I used your command line to create urdf and sdf files : 
`rosrun xacro xacro --inorder -o /home/ylo2/catkin_ws/src/quad-sdk/quad_simulator/ylo2_description/urdf/ylo2.urdf /home/ylo2/catkin_ws/src/quad-sdk/quad_simulator/ylo2_description/xacro/robot.xacro DEBUG:=false`

and

`gz sdf -p /home/ylo2/catkin_ws/src/quad-sdk/quad_simulator/ylo2_description/urdf/ylo2.urdf > /home/ylo2/catkin_ws/src/quad-sdk/quad_simulator/ylo2_description/sdf_mesh/ylo2.sdf`

elpimous, 2023-01-30:
I had an error on sdf file. this line was fault : I removed it !
[image]

when launching : `roslaunch quad_utils quad_gazebo.launch robot_type:=ylo2` I don't see any error on terminal.
[image]
[image]

elpimous, 2023-01-30:
Any idea ??

elpimous, 2023-01-31:
Well. SDF creation wasn't a success !!!
I corrected some things, starting from A1 original sdf file...
Now, working correctly.

[image]
Hope it will help each other.

closed 2023-01-31 by elpimous

DanielChaseButterfield, 2024-06-26:
@elpimous I'm having a very similar issue, I checked my sdf and urdf files, and they still don't work. You mentioned that you "I corrected some things, starting from A1 original sdf file..." Do you remember what you did?

Currently, I'm taking out the "ignition" line as you did earlier in this thread.

Also, I noticed that running `gz sdf` currently generates a sdf file with version=1.7 instead of 1.6 as before, so I wondered if that could be the cause. However, when I try to use `gz sdf` to generate a file with version=1.6, I get the following error:
```
Error [SDF.cc:459] Unable to find SDF filename[root.sdf] with version =1.7
Error [parser.cc:159] Could not find the 'element' element in the xml file
ERROR: SDF parsing the xml failed
```

elpimous, 2024-06-26:
Hi Daniel.
Building SDF file from command Line don t work for me.
I had to use a default one, for ex A1, and adapt it with my one. (I did the SDF build command line, only to be able to extract needed values, for mt A1 adapted SDF file.
A long process, but working.
Hope it will help you
Vincent