robomechanics/quad-sdk #341: real robot controller example?!

elpimous, 2023-01-31:
Hello all,
Do you have a controller example, working on a real robot?
Thanks

jcnorby, 2023-02-01:
Hey there! I'm guessing you are asking for instructions for running the software on robot hardware rather than in simulation.

First, a disclaimer - Quad-SDK is not guaranteed (or even designed) to work seamlessly with every quadruped hardware platform out there, since there are so many good ones these days! It would be difficult for us to support each of them, especially since many people like to have their own hardware setups/integrations. We made a specific decision to focus the software on what would apply to the largest portion of these robots (high and mid-level motion planning, low-level joint control), but once our system computes joint torques, it's somewhat up to the user to make sure they get to where they need to go.

That said, we did try to make this as easy as possible by both providing a) a simulator so folks could tinker with/extend the software without needing a hardware platform, and b) the abstract HardwareInterface class (as you mentioned) so that integrating with other robots would be as easy as possible. The derived HardwareInterface provided is for the Spirit 40 robot (the one we have in lab), but should serve as a good template for your particular platform.

In terms of actually launching the stack on a hardware platform, you can refer to [this page](https://github.com/robomechanics/quad-sdk/wiki/RML-Guide:-Running-Spirit) in the Wiki for an example for how we do it in our lab. Your mileage may vary, but our general approach was to launch high-rate nodes on the robot (e.g. NMPC and joint control, this is what is launched by robot_driver.launch), and slower-rate nodes on a remote computer (global planning, visualization, NMPC if you want it computed really fast).

Hope this helps!

elpimous, 2023-02-01:
@jcnorby what a good answer...
Thanks a lot....
Invertigating...

elpimous, 2023-02-01:
@jcnorby what a good answer...
Thanks a lot....
Invertigating...

1/ Can I / where can I change gait params?
2/ in one of your videos, the robot jumps, during walk... How did you do that? Stop walk - > jump - > restart walk?  Mix both controllers?
how did you do for jump : ask for certain body height?
Really exciting!!!

elpimous, 2023-02-02:
Another question, please : 
i'm working on my hardware_interface, (starting from spirit_interface base), with imu, send and rcvs functions.
How to test my hardware_interface ?
When testing robot_driver.launch with robot_type:=ylo2, it should start my hwinterface, correct ?  It asks for IP...
ps catkin build is ok, and ylo2_interface compiles seems to compile correctly. 
Could you tell me steps to start default controller, with my hardware_interface ?
Bests regards.
Vincent.

jcnorby, 2023-02-06:
Gait parameters are set in local_planner.yaml, specifically the period, duty_cycles, and phase_offsets params, which respectively set which period of the gait, the fraction of the period for which that leg is in stance, and the time within one gait cycle that the leg touches down. Legs and joint numbering is explained [here](https://github.com/robomechanics/quad-sdk/wiki/FAQ#what-is-the-convention-for-joint-and-leg-numbering-in-the-robotstate-message).

Leaping is handled seamlessly by altering the reference trajectory and contact schedule. The reference trajectory comes from the global planner (although you could hardcode your own leap if you wanted to override this), and the contact schedule is updated to add a period of full stance during the leaping and landing phases with no contact during flight (this is handled [here](https://github.com/robomechanics/quad-sdk/blob/b04d4f7658215da018b4843c98f49e50b1b339c6/local_planner/src/local_footstep_planner.cpp#L107)).

closed 2023-02-06 by jcnorby

jcnorby, 2023-02-06:
Closing this because I think the initial question was addressed, if not we can reopen (or make new issues as needed).

elpimous, 2023-02-06:
Thanks a lot @jcnorby