robomechanics/quad-sdk #345: NMPC solving fail when moving towards goal point

ujjdaniel, 2023-02-08:
Hi there !
I have an issue, that the planner messes up, when the robot is moving towards goal point in the middle of trajectory.
I uploaded my own map to test the Spirit robot moving around there but when I want him to reach goal state at the top of the ramp, he always has some problems from the middle of the ramp - to the top of ramp. It goes all well like on this picture 
[image]

but suddenly after going towards a little more, the footstep planner always gives him some nonsense values and points and it wanna go into flight phase, where the global planner is showing that there is NMPC solving fail. 
It looks like this
[image]

after that the robot falls down and it is not capable of standing up again and continuing the trajectory towards goal again.
[image]

When I am using twist input as reference instead of global planner, the robot doesnt have any problem to reach top of the ramp or moving around to any point on the map where I am sending him.
Can you help me figure out, where can be the problem ? Or at least in which part ? If it is most likely in global planning stack or it can be in local planner with footstep planner ?  Because the local planner works well when the twist reference is used, and global planner shows trajectories also well when I am using global planner reference so I dont know what can cause problems, that they dont work together well.

jcnorby, 2023-02-11:
Huh, that is quite strange, looks like you may have caught an actual bug! Is this behavior consistent, and if so does it always occur in the same location?

If everything is working correctly the local planner should only perform a leap if the global plan contains a section that is explicitly labeled as a leap, and from your visualization it doesn't look like that is the case (leap sections are denoted with different colors - green for leaping/landing phases and blue for flight phases).

One quick thing you could try is merging in [this commit](https://github.com/robomechanics/quad-sdk/pull/299/commits/113ee7dff6f56f7464b88210b8d4eaa39bd579b6) to see if that fixes this issue (this was recently pulled into devel so you could probably just merge that branch). That commit fixes an issue I found with updating the contact schedule when leaping. However I doubt that will fix it because it should only apply when the global plan contains a leap, which it shouldn't here.

jcnorby, 2023-02-11:
To help debug, could you add this code to the top of `LocalFootstepPlanner::computeContactSchedule()`?
```
for (auto primitive_id : ref_primitive_plan) {
  if (primitive_id != CONNECT_STANCE) {
    std::cout << "primitive_id = " << primitive_id << std::endl;
    // Add any other debug processing you want here
  }
}
```
This should print to the console if the reference plan contains any atypical primitive information - if it does then the issue is with the global planner doing something it shouldn't, but if it doesn't then the global plan is fine and the local planner is getting something wrong.

ujjdaniel, 2023-02-15:
@jcnorby thanks for your answer.
I tried to merge the commit you sent but it does not help to fix the issue. I also added the code you sent into the function but it doesn't print to console any atypical information, only few moments after the robot falls completely down. At the walking stage where it throws these nonsense trajectories it doesnt print to console any wrong information. 
To your question if it happens in the same location - on that map yes it does approximately in the same location on the map and every single time i try to run it. But it doesn't happen only at that map. It also happens on other custom maps where it doesn't go on ramp. Here is a picture for example from another terrain
[image]

So you say that the global planner should be fine, only the local planner of footstep planner is doing something wrong ?
Because when i navigate the robot through twist input as reference, then the robot is going well through all the maps i tried and it hasn't got problems.

jcnorby, 2023-02-19:
Apologies for taking a little while getting to this, its been a busy time! I probably won't be able to get to debugging this week and I'm not sure if any current RML members will get to it. If you want to debug yourself, I'd say it seems quite likely that something wrong is happening inside computeContactSchedule. That is the function that actually parses the plan and adds the leap to the contact schedule - I'm guessing something weird is happening there and you are just seeing the symptoms of that behavior. Other than that you could always inspect the data in the /global_plan topic (e.g. with a rostopic echo) and see if anything looks odd (either in the orientation data or primitive info).

jcnorby, 2023-02-26:
Ah, one additional possibility - could you print out the yaw reference trajectory (`std::cout << ref_body_plan_.col(5) << std::endl;`) at the end of LocalPlanner::addReference()? I wonder if these instances are occuring when yaw wraps around from -pi to pi. This is a known issue (see https://github.com/robomechanics/quad-sdk/issues/204) and one that has a pretty easy solution, we just haven't gotten around to it yet.

ujjdaniel, 2023-02-27:
@jcnorby this last possibility that you mentioned looks very promising. I tried it, and the robot with nmpc controller fails right at the time, when the yaw reference changes between -pi and +pi. I looked up to the issue #204 and i do not really understand the short term solution that you recommended there. Could you go more in depth about that or give me some pointers how to start/fix the issue? Thanks

AndrewZheng-1011, 2023-03-09:
Hey,

I also have run into the same issue and have tried unwrapping the yaw reference state that is fed into the MPC. Some quick changes were roughly made to get something:

**local_planner.h**
```
...
public:
  ...
  double prev_unwrapped_yaw;
  bool first_ref_plan;
...
```
**local_planner.cpp**
```
LocalPlanner::LocalPlanner(ros::NodeHandle nh) : local_body_planner_nonlinear_(), local_footstep_planner_() {
  ...
  first_ref_plan = true
}
...
void LocalPlanner::getReference() {
  ...
  } else {
    // Use global plan
    ...
    for (int i = 0; i < N_; i++) {
    ...
    }

    // Unwrap yaw angle for NMPC
    Eigen::VectorXd eig_unwrapped_yaw_ref = ref_body_plan_.col(5);
    
    std::vector<double> wrapped_yaw_ref, unwrapped_yaw_ref;
    quad_utils::eigenToVector(eig_unwrapped_yaw_ref, wrapped_yaw_ref);

    // If first plan, keep nominal yaw, else align initial yaw state with previous horizon's last yaw state 
    if (!first_ref_plan) {
      double diff = wrapped_yaw_ref[0] - prev_unwrapped_yaw;
      if (diff > M_PI) {
        wrapped_yaw_ref[0] = wrapped_yaw_ref[0] - 2*M_PI;
      } else if (diff < -M_PI) {
        wrapped_yaw_ref[0] = wrapped_yaw_ref[0] + 2*M_PI;
      }
      // std::cout << "new current yaw: " << wrapped_yaw_ref[0] << std::endl;
    } else {
      first_ref_plan = false;   
    } 
    unwrapped_yaw_ref = math_utils::unwrap(wrapped_yaw_ref); // Let math_utils handle unwrapping all yaw state in horizon
    prev_unwrapped_yaw = unwrapped_yaw_ref[N_ - 1];

    quad_utils::vectorToEigen(unwrapped_yaw_ref, eig_unwrapped_yaw_ref); // Convert to eigen

    // Insert unwrapped yaw into ref_body_plan_
    ref_body_plan_.col(5) = eig_unwrapped_yaw_ref;
    ...
  }
...
}
```

These quick code changes somewhat works out, but does not quite fix the issue. Here is a quick comparison through the gyf files:

_Wrapped Yaw Reference Trajectory_

[image]

_Unwrapped Yaw Reference Trajectory_

[image]

I am not quite sure what the issue is once I unwrapped the reference yaw trajectory. I will be looking into the output of the solver to see if it simply does not solve in time. 

There was mention of needing to change the warm start in issue #204 . I am not too familiar with CasADi's cpp warm start syntax and your paper mentions this as well. Could you perhaps elaborate on this matter? Perhaps that is the issue I see when I unwrapped the yaw reference.

AndrewZheng-1011, 2023-03-29:
Hey small update, unwrapping the reference yaw worked. However, I forgot to unwrap the current state as it was converted from quaternion to euler angles (giving a discontinuous domain). When I made those small changes, the NMPC somewhat works. However, it seems that the NMPC cannot go past $\pm \pi$ + $\epsilon$ when making these following changes, but at least it does not fail solving. 

I can make a more detailed report about this later, but I do believe it might be inherit to the general architecture. Example is when running a1 with teleop, I don't believe I can do 720 degree turns. Let me know if this might just be on my side.

closed 2023-05-01 by ardalantj