robomechanics/quad-sdk #346: Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from ...

elpimous, 2023-02-12:
hello the team.
My own hardware interface is ok, returning imu values into 0000
when launching hardware interface, I encounter this error : 
`Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
         at line 255 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp[ERROR] [1676208160.835078803, 0.349000000]: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from authority "unknown_publisher" because of an invalid quaternion in the transform (0,000000 0,000000 0,000000 0,000000)`

Any idea ?

elpimous, 2023-02-12:
some cout to test correct imu into quad controller : 
```
RECEIVE FUNCTION
Imu tests :
  Orientation x, y z et w = 0.601516 ; -0.382188 ; -0.610427 ; -0.345672
  angular_vel x, y and z = 0.112921 ; -0.0383507 ; -0.0330242
  linear accel x, y and z = -10.0224 ; 0.545895 ; -0.277736
```

elpimous, 2023-02-12:
the imu msg is feeded into robot_driver.cpp
https://github.com/elpimous/quad-sdk-ylo2-real-robot/blob/ce5d2ea2cf0231c205abf8ab46b2a1ee0fb98156/robot_driver/src/robot_driver.cpp#L294-L297

elpimous, 2023-02-12:
terminal example : 
`RECEIVE FUNCTION
Test de l'IMU :
  Orientation en x = 0.63549
RECEIVE FUNCTION
RECEIVE FUNCTION
Test de l'IMU :
  Orientation en x = 0.635452
RECEIVE FUNCTION
RECEIVE FUNCTION
Test de l'IMU :
  Orientation en x = 0.635415
RECEIVE FUNCTION
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
         at line 255 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
[ERROR] [1676210376.270850425, 1.670000000]: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from authority "unknown_publisher" because of an invalid quaternion in the transform (0,000000 0,000000 0,000000 0,000000)`

elpimous, 2023-02-12:
imu message from imu/data topic : 
`---
header: 
  seq: 2115
  stamp: 
    secs: 19
    nsecs: 223000000
  frame_id: "trunk"
orientation: 
  x: 0.21656912658068272
  y: 0.23204853624913963
  z: 0.3615810592380619
  w: 0.876647265345072
orientation_covariance: [4.592449e-06, 0.0, 0.0, 0.0, 4.592449e-06, 0.0, 0.0, 0.0, 4.592449e-06]
angular_velocity: 
  x: -0.01704475112515753
  y: -0.03835068899130714
  z: -0.036220097701405704
angular_velocity_covariance: [5.895184e-06, 0.0, 0.0, 0.0, 5.895184e-06, 0.0, 0.0, 0.0, 5.895184e-06]
linear_acceleration: 
  x: -2.3990632748432454
  y: 5.329655392593145
  z: 7.661679121255874
linear_acceleration_covariance: [0.0007199025610000001, 0.0, 0.0, 0.0, 0.0007199025610000001, 0.0, 0.0, 0.0, 0.0007199025610000001]
---

`

elpimous, 2023-02-12:
the frame of robot, when running : 
[frames.pdf](https://github.com/robomechanics/quad-sdk/files/10716425/frames.pdf)

elpimous, 2023-02-16:
Small Up, friends, please...I'm lost

jrenaf, 2023-02-21:
Hello @elpimous, this is a typical namespace issue. Please check you namespace in launch file and make sure the frame_id is in the format of "robot_1/ground_truth/body" rather than "robot_1_ground_truth/body".

elpimous, 2023-02-21:
Thanks @jrenaf .
I'll check it

closed 2023-02-23 by ardalantj

jcnorby, 2023-03-08:
Quick correction here - robot_1_ground_truth/body IS in fact the correct child_frame_id for the TF transform. TF only allows one namespace (not nested names like ROS typically allows), so the underscore rather than forward slash is correct. That said, the error here is informative - some TF you are trying to broadcast is not being initialized. Given the context of your post in https://github.com/robomechanics/quad-sdk/issues/350, I would guess it is due to an incorrect launching of nodes.