robomechanics/quad-sdk #347: /robot_1/state/ground_truth does not publishes contact_state

cmburgul, 2023-02-19:
I see the ground truth topic does not publishes contact state. The topic /robot_1/state/grfs do publish but the publish timestamps are not synchronized. Is there a way to synchronize the timestamps of grfs and ground_truth topic.

jcnorby, 2023-03-08:
These two data streams come from different plugins which run at different rates (a contact sensor and a model plugin) so they aren't automatically synced. Contact sensors only publish data individually over topics, so we have a contact_state_publisher node which subscribes to each of these and publishes a consolidated GRF message. If you wanted to sync them you could perhaps put that logic directly into the estimator plugin (add subscribers to the individual contact topics, then bundle everything together).

ardalantj, 2023-05-01:
Closing due to inactivity.

closed 2023-05-01 by ardalantj