robomechanics/quad-sdk #350: Error: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "robot_1_ground_truth/body" from ...

elpimous, 2023-03-03:
hello, regarding to

elpimous, 2023-03-07:
well, new test : 
- I used default noetic branch,
- kept all actual code,
- modified **send()** and **recv()** functions, witn my send/receive code.
- send and receive joints lib functions tested and working.
- imu working, feeding last_imu_msg_ (**imu_pub_.publish(last_imu_msg_) is tested, working**)

jcnorby, 2023-03-08:
To be clear, quad_gazebo.launch is for launching the robot on gazebo, not on hardware. We do not provide a readymade hardware launch file because every robot is different (as mentioned in https://github.com/robomechanics/quad-sdk/issues/341). I would suspect that by launching gazebo, somehow the broadcaster responsible for this particular transform thinks it has enough information to broadcast when in fact it does not. Also see my latest comment about namespacing in https://github.com/robomechanics/quad-sdk/issues/346.

So in summary - revert the namespacing change and use robot_driver.launch as a guide for how to write your own launch file (rather than quad_gazebo.launch). You could also take a look at the [JointController](https://github.com/robomechanics/quad-sdk/blob/main/robot_driver/src/controllers/joint_controller.cpp) class, which just controls individual joints and shouldn't need any state estimation (hardware state estimation is still a work in progress). That would help with bring up to make sure you are talking to joints correctly.

elpimous, 2023-03-08:
Oh, how can I be so stupid, Joe !!
I did not call the robot_driver.launch!!!

Now the robot gets up...at least, try, lol

Still some reversals of engines to operate and the dream is within my reach... 

Many thanks, @jcnorby  and see you soon in another question

closed 2023-03-09 by elpimous