robomechanics/quad-sdk #353: real robot standup back push

elpimous, 2023-03-09:
Hello the team.
Thanks for help.
Now, i can make my ylo2 robot move for a standup.
URDF ans SDF ok, inertias and gravities ok.

When running robot_hardware.launch, the robot automatically standup (after a homemade zero calibration = ok)
During standup, all joints have correct direction, imu is ok,

But robot is pushed backward : it stands up, but anything like 45° up-rear.
Any idea ?

elpimous, 2023-03-09:
https://youtu.be/LNuAhMMIKZ0

elpimous, 2023-03-10:
hello, well even with correct servos direction and correct sit down and standup params : 
```
ylo2@ylo2-UP-WHL01:~$ rosparam get /robot_driver/sit_joint_angles

- 0.165
- 0.59
- 1.36

ylo2@ylo2-UP-WHL01:~$ rosparam get /robot_driver/stand_joint_angles
- 0.0
- 0.5213467868148791
- 1.09094730619712
```
The robot put it legs right, and perpendicular to body (my real rezero joints)

I don't anderstand what happens..  Any small idea to share to help me progress ?!

elpimous, 2023-03-11:
Here is my steps : 
Ylo2 robot is permanent zeroed with legs parallel, right, and perpendicular to body
[image]

Before running code, robot is in a sit position : 
[image]

elpimous, 2023-03-12:
Or perhaps in hardware mode, initial pose and stand are set in other place?  If yes, didn't find...

jrenaf, 2023-03-13:
Hello @elpimous, it's great to see you are trying quad-sdk on new quadruped platform! The stand-up motion is hardcoded in robot_driver and it might be impacted by different initialization configuration. You might directly change that part [here](https://github.com/robomechanics/quad-sdk/blob/main/robot_driver/src/robot_driver.cpp#L418) .

elpimous, 2023-03-13:
Hi @jrenaf
Yes I saw that.
My question is : does the hardware use sit and params, from quad_util/config/ylo2_ config.yaml

And how is calibrate spirit?

elpimous, 2023-03-14:
Hey, i can confirm the following things : 
- any changes on **sit_joint_angles** param, in a1.yaml ( https://github.com/robomechanics/quad-sdk/blob/b04d4f7658215da018b4843c98f49e50b1b339c6/quad_utils/config/a1.yaml#LL76C2-L76C36 )
are without effect on hardware.

(changes are ok on a1 simulation )

My urdf is a1, and i call robot_type = a1.

elpimous, 2023-03-15:
Hello the team.
Found an error in my joints order...
Working now on joint direction, and initial pose.. Next soon.

closed 2023-03-15 by elpimous