robomechanics/quad-sdk #354: Questions about tuning "my_robot".yaml file, for real working robot

elpimous, 2023-03-16:
Hi all.
1/ When launching `robot_driver.launch` , legs move to their initial pose, thanks to param : `sit_joint_angles`
During this process, the variable **control_mode_** takes the value : SIT (0)

2/ I send command for a standup : `rostopic pub -1 /control/mode std_msgs/UInt8 "data: 1"`

jrenaf, 2023-03-18:
Hello @elpimous, you can tune the PD through visualization of motor position by letting live_plot to be true, which will open PlotJuggler that can show the instantenous position tracking by each motor.

elpimous, 2023-03-18:
@jrenaf  "Do you have any idea how you want to do estimation on your robot in the future?"
Sorry, don't anderstand your question . Would you mind explain me ?

Thanks for tips !

### Here was my PID steps : 
1 - put **kd** to 0
2 - find best **kp** value : highter is better, until i see motors trembling, then I lower a bit to erase any tremblings.
3 - let **kp** at it best, and do same with **kd**, until i see best response, and no tremblings.
actually, for now, I'm at this point : 

```
sit_kp: [1, 1, 1]
sit_kd: [0.1, 0.1, 0.1]
stand_kp: [3.5, 3.5, 3.5]
stand_kd: [1.3, 1.3, 1.3]
```

closed 2023-03-22 by elpimous