robomechanics/quad-sdk #355: Command to make real robot walk.

elpimous, 2023-03-18:
Hi the team.
Well, the controller, adapted to my ylo2 robot takes form, cool.
It can standup and sitdown correctly (thanks for your help)

Now, I'm trying to make it walk, using same commands as in simulation : 

roslaunch quad_utils quad_plan.launch reference:=twist logging:=true
rosrun teleop_twist_keyboard teleop_twist_keyboard.py cmd_vel:=/robot_1/cmd_vel

Is there an easy way to add commands on robot_driver.launch , to be able to play with /cmd_vel topic ?

Thanks the team.

jrenaf, 2023-03-19:
Hi @elpimous, what kind of commands do you want to add? The velocity commands are actually listened by the local planner not the robot driver. Also, the local planner needs to know its body pose, which in our case is given by either motion capture or the onboard tracking camera, I don't know how your robot estimates its own body pose.

elpimous, 2023-03-20:
Hey Jim,  I don't use any motion capture.

@jrenaf , here is my robot_driver.launch file : 

```
<launch>
    <arg name="robot_type" default="a1"/>
    <arg name="live_plot" default="false" />
    <arg name="dash" default="false" />
    <arg name="logging" default="false" />
    <arg name="controller" default="inverse_dynamics" />
    <arg name="is_hardware" default="true" />
    <arg name="namespace"       default="robot_1" />

    <rosparam command="load" file="$(find global_body_planner)/global_body_planner.yaml" />
    <rosparam command="load" file="$(find local_planner)/local_planner.yaml" />
    <rosparam command="load" file="$(find nmpc_controller)/nmpc_controller.yaml" />
    <rosparam command="load" file="$(find robot_driver)/robot_driver.yaml" />

    <include file="$(find quad_utils)/launch/load_robot_params.launch">
      <arg name="robot_type" value="$(arg robot_type)"/>
    </include>

    <!-- Launch the robot driver node -->
    <node name="robot_driver" pkg="robot_driver" type="robot_driver_node" output="screen">
        <param name="controller" value="$(arg controller)" />
        <param name="is_hardware" value="$(arg is_hardware)" />
    </node>

</launch>
```

I launch only this node.
I don't have any motion capture.

In simulation, i can make robot move with those commands : 

```
 roslaunch quad_utils quad_plan.launch reference:=twist logging:=true
rosrun teleop_twist_keyboard teleop_twist_keyboard.py cmd_vel:=/robot_1/cmd_vel
```

My question is : what commands or launch must I add in my robot_driver.launch, and what terminal command must I run to make the robot walk ?

Thanks friend.

elpimous, 2023-03-20:
When yoy say motion capture,
Are you talking about camera motion capture,

Or robot motion capture : robot estimates it pose with its movements.

Here is perhaps my bad language conversion

elpimous, 2023-03-20:
I’m afraid I can’t make my robot work without a motion capture camera or tracking camera?

closed 2023-03-22 by elpimous