robomechanics/quad-sdk #356: how to estimate robot body pose, without camera(s) Help !

elpimous, 2023-03-20:
Hi friends.
quad-sdk estimates body pose with help of cameras..
Did anyone here did it without camera ?
Could I ask any help, from a "newbie", to help my robot walk ?
Thanks all.
Vincent

ologandavid, 2023-03-20:
Do you mean just using the simulation ground truth for your body pose?

elpimous, 2023-03-20:
hi big thanks for help.
Well, don't know ! 
Using code on my real robot.
It stand up correctly, but i'd like to make it walk only with the robot itself, without aly mocap or cameras.
All ideas would help me a lot

ologandavid, 2023-03-20:
@elpimous Does your robot currently have an IMU? Assuming the robot doesnt slip, you could integrate the IMU to get a position and velocity estimate. This might drift, but if you're impatient, this might be your best bet.

elpimous, 2023-03-20:
PS I have a t265, place on robot bottom (d435 front) 
I don't see in code how to use t265 ?!!

ologandavid, 2023-03-20:
That sounds right, feeding in your IMU information into last_robot_state_msg should work. I'll keep you updated on the progress of EKF estimation!

elpimous, 2023-03-22:
Didn't find it.
Discussion section, check

closed 2023-03-22 by elpimous