robomechanics/quad-sdk #357: How to use T265 on controller.?

elpimous, 2023-03-22:
Hi the team.
Well, I have a T265 cam. I'd like to use it with local Planner.
On the launch, I set false to mocap, and I run the T265 launch.
The T265 is in place into urdf... 

What more to do to to use it with quad SDK?
Big thanks, all.
vincent

elpimous, 2023-03-23:
Hello. Found 
`https://github.com/robomechanics/quad-sdk/commit/f4733587e34d979f87a0210b90a60a8a74a11082`
It seems to integrate the T265.
After the T265 launch and the robot driver launch,
the T265 topic should replace the mocap one, correct?
Don't know where to do it..
😭😅

jrenaf, 2023-03-24:
Hello @elpimous, we haven't integrated T265 stuff into main branch. But one way to go is trying out melodic_devel_go_outdoor branch. This branch has no multi-robot support though, so no need to add robot_1 namespace in roslaunch.

elpimous, 2023-03-24:
Thanks, friend.
@jrenaf  trying and posting here results.. 👍

jrenaf, 2023-03-26:
The fourth step should still be `roslaunch quad_utils remote_driver.launch `. There is one more step between 4 and 5: `roslaunch quad_utils planning.launch `

elpimous, 2023-03-26:
hi @jrenaf  really happy ro read you.
I don't use any remote control, for now (i'll study it launches)
thanks for step mpatient to test.
I'll tell you 
Thanks a lot @jrenaf

elpimous, 2023-03-26:
tested : 
running : 
```
roslaunch quad_utils robot_driver.launch
roslaunch quad_utils t265.launch
roslaunch quad_utils remote_driver.launch
roslaunch quad_utils planning.launch
```

and standing robot : 
```
rostopic pub -1 /control/mode std_msgs/UInt8 "data: 1" 
publishing and latching message for 3.0 seconds
```

terminal command is : 

```
[ INFO] [1679868865.694767562]: GBP has state and map information
[ INFO] [1679868865.695595708]: In reset mode
[ WARN] [1679868865.695651408]: Invalid start state, exiting
[ WARN] [1679868866.695776348]: Invalid start state, exiting
[ INFO] [1679868867.695770153]: In reset mode
[ WARN] [1679868867.695805279]: Invalid start state, exiting
[ WARN] [1679868868.745788318]: Invalid start state, exiting
[ INFO] [1679868869.695801764]: In reset mode
...
```

and when running : `rosrun teleop_twist_keyboard teleop_twist_keyboard.py cmd_vel:=/robot_1/cmd_vel`

terminal command is : 

Waiting for subscriber to connect to /robot_1/cmd_vel

Did I do anything wrong ?

elpimous, 2023-03-26:
Why do I need remote_driver.launch with no real remote  ?

elpimous, 2023-03-27:
[image]

elpimous, 2023-03-27:
Should the robot automatically walk just after planning.launch?
Or legs moves come from a bad robot in rviz?!

jrenaf, 2023-03-28:
Oh, two things you might need to adjust that I forget to mention: first, comment out line 557 of `quad_utils/src/rviz_interface.cpp` and second, you might need to change the transformation inside t265.launch based on where you set `camera_odom_frame` and body CoM on your own robot. Note that +x is pointing forward, +y leftward and +z upward.

elpimous, 2023-03-28:
Thanks..
Yes actually I commented this line
ground_truth_base_tf_br_.sendTransform(transformStamped);
To
// ground_truth_base_tf_br_.sendTransform(transformStamped);

Ok for transforms

jrenaf, 2023-03-29:
Hi @elpimous, you saw that ground truth error on the left column of Rviz? That's why this error happened. A quick check would be running rqt_tf_tree and see whether all transformation are connected correctly. It would be nice if you could pose a screenshot of your tf tree.

elpimous, 2023-03-29:
Hello @jrenaf
Here is the rqt tree :

elpimous, 2023-03-29:
[image]
The rviz error, when all nodes running.

elpimous, 2023-03-31:
hello @jrenaf , any news ?

elpimous, 2023-04-01:
[image]

elpimous, 2023-04-01:
[image]

elpimous, 2023-04-01:
[image]

elpimous, 2023-04-03:
Hey, @jrenaf 
Jiming, could you help in the error ? 
Have a nice day
Vincent

jrenaf, 2023-04-03:
Hello @elpimous, sorry for the late reponse. I was really busy in the past week. Your rqt_tf_tree looks right so it's wierd why the transformation shows error. Can you expand Ground Truth State term and see what's wrong inside?

elpimous, 2023-04-03:
Hey, Jiming,
Not problem, of course.
you're asking : ### Can you expand Ground Truth State term and see what's wrong inside? 
not sure to anderstand what you want ?!
Thanks Jiming

elpimous, 2023-04-03:
[image]
[image]

jrenaf, 2023-04-03:
Oh I see, you need to get rid of the slash before all the frames. In Your case, it should be "groud_truth/body" not "/ground_truth/body".

elpimous, 2023-04-03:
Hello @jrenaf
Well , looked in remote_driver.launch and all others launches from quad_utils.
I don't see any location with `/ground_truth/body` to change to ground_truth/body... 
If you remember where it was, i would be enjoyed lol

elpimous, 2023-04-04:
Hi @jrenaf 
Doesn't change anything !
tested, anyway, a `roslaunch quad_utils planning.launch`
_content of it terminal_ : 
```
[ WARN] [1680630880.094511957]: No cmd_vel data, setting twist cmd_vel to zero
[ INFO] [1680630880.114032588]: LocalPlanner took 19.458 ms
[ WARN] [1680630880.280990115]: NMPC solving fail
[ INFO] [1680630880.300736053]: LocalPlanner took 19.632 ms
[ INFO] [1680630880.498606450]: LocalPlanner took 103.073 ms
[ WARN] [1680630880.622697779]: NMPC solving fail
[ INFO] [1680630880.644465306]: LocalPlanner took 21.680 ms
[ INFO] [1680630880.751951425]: LocalPlanner took 52.171 ms
[ WARN] [1680630880.873234071]: NMPC solving fail
[ WARN] [1680630880.993977814]: NMPC solving fail
[ INFO] [1680630881.015463140]: LocalPlanner took 21.388 ms
[ INFO] [1680630881.119693976]: LocalPlanner took 54.543 ms
```

then a : `rostopic pub /control/mode std_msgs/UInt8 "data: 1"`
_content of it terminal_ : 
```
[ INFO] [1680630897.089804499]: LocalPlanner took 14.298 ms
[ WARN] [1680630897.154131013]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.161748262]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.173355963]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.185970656]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.186023936]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ INFO] [1680630897.197828026]: LocalPlanner took 11.949 ms
[ WARN] [1680630897.199040005]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.211313569]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.218547770]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.240188445]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.244704359]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.249051412]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.253321909]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.253371329]: computeFootPlan prediction receives a position out of range, pick the previous position in map!
[ WARN] [1680630897.375740789]: NMPC solving fail
[ INFO] [1680630897.402590297]: LocalPlanner took 26.758 ms
```
The legs were out of control. !

I think that removing the slash is a good start... Searching again.

jrenaf, 2023-04-04:
Is the robot showed on rviz?

elpimous, 2023-04-04:
Hi @jrenaf
Well, changed some /ground_truth" to "ground_truth"...
Now, I see full correct robot on rviz, cool
[image]

elpimous, 2023-04-04:
now, i can see i forgot to modify my robot specs... size... (legs position are strange, however !)
Any idea about the local_plan_position error (on rviz ?)

tested a planning.launch ..... BAD !! Legs were out of control, lol

elpimous, 2023-04-05:
[image]
[image]

elpimous, 2023-04-05:
[image]
[image]

jrenaf, 2023-04-07:
Sorry again for the late reply, another busy week just had.The robot showing up is a really good sign. But the leg inverted is definitely not right and we don't have that issue on Spirit (but I don't know why it stands up correctly). You said the robot freaked out after planning, one thing to do when robot sitting on the ground is to lift up the robot by hands and waggle a little bit to see whether the robot estimate is correct inside Rviz.

elpimous, 2023-04-07:
Hi @jrenaf 
Happy to read you.
I think i know the leg problem !!
my real robot is zero calibrated with legs parallel to body, and extended.
your robot seems to be zeroed with lower leg bent inside upper leg.
So, my robot lower leg start it angle from full extension.
Your mind ?
An idea about the local_plan_position red error on rviz ?
See you soon, Jiming
Vincent

elpimous, 2023-04-08:
Hey, Jiming.
@jrenaf any news : 
after some mods on urdf, legs appears correctly, and move in same order than when I move them with hand. cool.
Now, when I standup robot, i can see thar robot stand on ground, legs extend outside of ground, and t265 axis go up !
[image]

[image]

I suspect a parent/child error !  searching...
If you have answer before me....... Thanks.
Vincent

jrenaf, 2023-04-08:
That looks like a error of transformation between camera_odom to body CoM. Maybe should double check T265.launch.

elpimous, 2023-04-08:
hello @jrenaf 
well, nothing about body COM under t265.launch..

```
  <node pkg="tf2_ros" type="static_transform_publisher" name="tf_map_to_cam_odom_broadcaster" 
        args="0.109 0 0.189 0 0 0 1 map camera_odom_frame" />
  <node pkg="tf2_ros" type="static_transform_publisher" name="tf_cam_pose_to_body_broadcaster" 
        args="-0.109 0 -0.135 0 0 0 1 camera_pose_frame ground_truth/body" />

  <node name="camera_pose_frame_tf_listener" pkg="quad_utils" type="tf_to_pose_publisher.py" output="screen"/>
</launch>
```

searching...

jrenaf, 2023-04-09:
Can I take a look at your rqt_tf_tree again?

elpimous, 2023-04-09:
[image]
Hey, here it is.

jrenaf, 2023-04-10:
Looks good. And can I take a look at rqt_graph as well?

elpimous, 2023-04-10:
Hi @jrenaf 
here it is : 
[image]

elpimous, 2023-04-10:
When sending a standup, I can see robot extend its legs correctly, see the t265 imu axis going up, but body robot stays on the ground

jrenaf, 2023-04-10:
Can you untick all the boxes in Hide on rqt_graph?

elpimous, 2023-04-10:
here it is unticked hide, and unticked all
[image]
[image]

jrenaf, 2023-04-10:
Ok, I think every node's name here has a slash in the front, which is the same issue as tf_tree.

elpimous, 2023-04-10:
I need to remove all slash in front of each topic, correct ?
I was on the branch.
Waiting for your answer.
Thanks

elpimous, 2023-06-19:
hi friends , did you ever seen this error : 
into rviz, red error : 
**Error subscribing: Tried to subscribe to a topic with the same name but different md5sum as a topic that was already subscribed [geometry_msgs/PoseArray/916c28c5764443f268b296bb671b9d97 vs. visualization_msgs/Marker/4048c9de2a16f4ae8e0538085ebf1b97]**
[image]

ardalantj, 2023-06-24:
Hi @elpimous thanks for being very active and using Quad-SDK. While we would love to help as many Quad-SDK users as possible, it would be very difficult for our team to solve every problem for every user due to limited bandwidth. To be able to support you and other users better, I suggest keeping each issue focused on a single problem at a time and post issues on Github for problems that you are really stuck on after trying different solutions. This way our team can focus on addressing critical problems for users and better allocate resources. Thanks again for your continued engagement and we look forward to see your robot integrated with Quad-SDK.

ardalantj, 2023-07-17:
Closing due to inactivity.

closed 2023-07-17 by ardalantj