robomechanics/quad-sdk #359: all "outdoor" branches doesn't work correctly

elpimous, 2023-04-03:
Hi the team,
tested both "outdoor" branches : 

**branch : melodic_devel_go_outdoors**
```
ylo2@ylo2-ROG-Strix-G713QR-G713QR:~/catkin_ws$ source devel/setup.bash
ylo2@ylo2-ROG-Strix-G713QR-G713QR:~/catkin_ws$ roslaunch quad_utils quad_gazebo.launch
... logging to /home/ylo2/.ros/log/60557200-d23e-11ed-976d-bf396cdebb59/roslaunch-ylo2-ROG-Strix-G713QR-G713QR-89983.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ylo2-ROG-Strix-G713QR-G713QR:40895/

SUMMARY
========

PARAMETERS
 * /body_force_estimator/update_rate_: 200
 * /contact_detection/update_rate: 250
 * /contact_state_publisher/update_rate: 500
 * /ekf_estimator/bias_x: 0.08
 * /ekf_estimator/bias_y: 0.06
 * /ekf_estimator/bias_z: -9.800917657
 * /ekf_estimator/joint_state_max_time: 20
 * /ekf_estimator/update_rate: 100
 * /enable_statistics: True
 * /gazebo/enable_ros_network: True
 * /global_body_planner/BACKUP_RATIO: 0.5
 * /global_body_planner/BACKUP_TIME: 0.3
 * /global_body_planner/BODY_TRAVERSABILITY_THRESHOLD: 0.75
 * /global_body_planner/CONTACT_TRAVERSABILITY_THRESHOLD: 0.01
 * /global_body_planner/DY_MAX: 2.0
 * /global_body_planner/F_MAX: 250.0
 * /global_body_planner/F_MIN: 0.0
 * /global_body_planner/GOAL_BOUNDS: 0.5
 * /global_body_planner/G_CONST: 9.81
 * /global_body_planner/H_MAX: 0.375
 * /global_body_planner/H_MIN: 0.1
 * /global_body_planner/H_NOM: 0.3
 * /global_body_planner/J_CONST: 1.0
 * /global_body_planner/KINEMATICS_RES: 0.03
 * /global_body_planner/MU: 0.25
 * /global_body_planner/M_CONST: 13.0
 * /global_body_planner/NUM_LEAP_SAMPLES: 10
 * /global_body_planner/PEAK_GRF_MAX: 7.0
 * /global_body_planner/PEAK_GRF_MIN: 4.0
 * /global_body_planner/ROBOT_H: 0.05
 * /global_body_planner/ROBOT_L: 0.4
 * /global_body_planner/ROBOT_W: 0.3
 * /global_body_planner/TRAPPED_BUFFER_FACTOR: 4
 * /global_body_planner/T_F_MAX: 0.4
 * /global_body_planner/T_F_MIN: 0.1
 * /global_body_planner/T_S_MAX: 0.3
 * /global_body_planner/T_S_MIN: 0.3
 * /global_body_planner/V_MAX: 2.0
 * /global_body_planner/V_NOM: 0.5
 * /global_body_planner/goal_state: [6.0, 0.0, 0.3, 0...
 * /global_body_planner/max_planning_time: 0.25
 * /global_body_planner/num_calls: 1
 * /global_body_planner/replanning: True
 * /global_body_planner/start_state: [0.0, 0.0, 0.3, 0...
 * /global_body_planner/startup_delay: 1.5
 * /global_body_planner/state_error_threshold: 25
 * /global_body_planner/update_rate: 20
 * /grid_map_filter_node/grid_map_filters: [{'name': 'duplic...
 * /grid_map_filter_node/input_topic: /terrain_map_raw
 * /grid_map_filter_node/output_topic: /terrain_map
 * /grid_map_visualization/grid_map_topic: /terrain_map
 * /grid_map_visualization/grid_map_visualizations: [{'name': 'z_poin...
[... 271 lines cut ...]
[ INFO] [1680540088.501052633, 0.368000000]: <topicName> set to: /state/imu
[ INFO] [1680540088.501066045, 0.368000000]: <frameName> set to: imu::link
[ INFO] [1680540088.501093637, 0.368000000]: <updateRateHZ> set to: 500
[ INFO] [1680540088.501107537, 0.368000000]: <gaussianNoise> set to: 0
[ INFO] [1680540088.501129471, 0.368000000]: <xyzOffset> set to: 0 0 0
[ INFO] [1680540088.501154688, 0.368000000]: <rpyOffset> set to: 0 -0 0
[ INFO] [1680540088.618846454, 0.368000000]: Loading gazebo_ros_control plugin
[ WARN] [1680540088.618933701, 0.368000000]: Desired controller update period (0.002000000 s) is slower than the gazebo simulation period (0
[ INFO] [1680540088.618971422, 0.368000000]: Starting gazebo_ros_control plugin in namespace: /spirit
[ INFO] [1680540088.619397106, 0.368000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the RO
[ INFO] [1680540088.743945438, 0.368000000]: Loaded gazebo_ros_control.
[INFO] [1680540088.765195, 0.368000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1680540088.766793, 0.368000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1680540088.768064, 0.368000]: Loading controller: joint_controller
[ INFO] [1680540088.994562034, 0.368000000]: Ground Truth State Estimator: <updateRateHZ> set to: 500
[INFO] [1680540089.002749, 0.373000]: Loading controller: joint_state_controller
[INFO] [1680540089.010276, 0.379000]: Controller Spawner: Loaded controllers: joint_controller, joint_state_controller
[INFO] [1680540089.012943, 0.381000]: Started controllers: joint_controller, joint_state_controller
[spawn_sdf_model-11] process has finished cleanly
log file: /home/ylo2/.ros/log/60557200-d23e-11ed-976d-bf396cdebb59/spawn_sdf_model-11*.log
```
[image]

doesn't work (simulation or hardware produce bad robot into rviz, and errors when hardware)
Any ideas ? (same error in t265 issue : )

closed 2023-04-21 by jrenaf

Cdfghglz, 2023-05-18:
Could you please share the solution to the problem above?

elpimous, 2023-09-03:
@Cdfghglz I don't have success for now.. sorry