robomechanics/quad-sdk #364: how to rest real imu ?

elpimous, 2023-04-26:
Hi the team,
Well, using melodic_devel_go_outdoors to test walk, helped with t265 camera.
I use too an imu, publishing to `imu/data topic`.

the imu is coded into hardware_interface, and publish correctly into last_imu_msg_ variable.

**Question 1** : when moving real robot by hand, i should see rviz one moving too, correct ?
**Question 2** : i don't see in your code what parent/child/frame i should use for my imu ? Could you help in this setup ?

note : I use a myahrs+ imu, with a separate launch. It's not a unitree imu, so for that, i'm a bit lost !

Thanks in advance, the team, or anyone 

Vincent

elpimous, 2023-04-28:
Hey friends, Any help, please ?

jcnorby, 2023-05-08:
I may be able to help briefly, although these sorts of issues are best debugged by just delving into the code and figuring out how things are connected. In this particular case, the robot model you see in RViz is visualizing data from /tf, so you could try making sure your TF tree looks complete (connections from the map frame to all the link frames).

The TF data is published in two ways. First, the map -> body transform is published by rviz_interface, using data from the /state/XXX topics (where XXX is usually ground_truth). Second the body -> link TFs are published by the robot_state_publisher node (launched [here](https://github.com/robomechanics/quad-sdk/blob/main/quad_utils/launch/visualization_plugins.launch)), which takes in joint data via a topic and broadcasts the appropriate TFs. In this case, rviz_interface also parses the robot state topic and rebroadcasts the appropriate joint_state topic so that robot_state_publisher can do its thing.

All this means there are a few areas you should inspect for where things have gone wrong. Most likely the robot state topic is not getting the imu data properly populated, and so none of the rest of the system has the right data. However, it you can verify that topic is correctly populated you could try checking these other locations (rviz_interface and the topics it publishes). This functionality is pretty well tested but depending on what custom modifications you've made its hard to say what exactly is the issue.

elpimous, 2023-05-08:
@jcnorby  thanks a lot for this big answer.
Ill check that and I'll feed result. 
Thanks

ardalantj, 2023-05-08:
Are there any followups here or can this be closed?

closed 2023-05-08 by elpimous