robomechanics/quad-sdk #367: A problem loading my own map file

hrxsd, 2023-05-08:
Hello developers, I am sorry to disturb you. I have a question about adding my own terrain maps built in solidworks.
When I add my own map file according to the instructions in the WIKI, the map is not displayed in the rviz interface, and the terminal reports an error as
[mesh_to_grid_map_node-2] process has died [pid 13961, exit code -11, cmd /home/m/catkin_ws_quadSDK/devel/lib/quad_utils/mesh_to_grid_map_node __name:=mesh_to_grid_map_node __log:=/home/m/.ros/log/e5a279fa-ed58-11ed-9170-90de804d6270/mesh_to_grid_map_node-2.log].
log file: /home/m/.ros/log/e5a279fa-ed58-11ed-9170-90de804d6270/mesh_to_grid_map_node-2*.log
Please help me solve this problem, thank you very much!
[image]

hrxsd, 2023-05-08:
I think I have solved the above problem by fixing the coordinate information in solidworks. But now there is a new problem: the initial position of the robot is not correct. Which parameters in which files should I change?
[image]

ologandavid, 2023-05-08:
Hi @hrxsd,
To correct the initial pose of the robot in sim, you would have to edit the launch file itself. Inside quad_utils, line 37 of quad_gazebo.launch lets you specify the inital pose of the robot. Correct it here to the appropriate pose, and see if that resolves your issue.

hrxsd, 2023-05-09:
Thank you for answering this issue! @ologandavid ,
After I modified the initial_pose in the method you provided, the robot does show the appropriate initial position in the rviz. But when I got the robot up and running, the robot fell through the map model. So I presume the root cause is still in the map file I built.
Also, if the parameters in the launch file are changed it means that the initial position of the robot in other worlds has changed. So I will try to fix my map file so that the coordinates of the map align with the coordinates required by the launch file.
[image]

hrxsd, 2023-05-09:
Thanks again, I have solved the problem by aligning the coordinates and I will close this issue.

closed 2023-05-09 by hrxsd

nanbwrn, 2023-07-20:
@hrxsd 
Could you please briefly explain the steps you took to solve this issue? I think I might have encountered a similar problem.
[image]

ruihuang1124, 2024-04-23:
> Thanks again, I have solved the problem by aligning the coordinates and I will close this issue.

Hi friend @hrxsd , I encounter a similar problem, can you briefly explain how you align the coordinates?

ssaigarimella, 2024-08-29:
How do I edit the worlds? Like for example I want to make it longer by 5 times so the robot can walk for a longer distance, how do I do this? @ologandavid @hrxsd

ologandavid, 2024-08-29:
@ssaigarimella, 
I would first recommend referring to the instructions provided here: https://github.com/robomechanics/quad-sdk/wiki/Tutorial:-Creating-Custom-Terrain-Map-Files, which should be enough to get you started. 

Best, 
David

ssaigarimella, 2024-08-30:
@ologandavid thanks for your reply. I created the world per the tutorial. It shows up correctly in gazebo, but not in the rviz and therefore the local planner fails. Any suggestions on this? thanks

ssaigarimella, 2024-09-01:
> dinates and I will close this issue.

@hrxsd What do you mean by fixing the coordinate information in solidworks? I am having the same issue where I cant view the mesh in rviz