robomechanics/quad-sdk #369: how to install quad-sdk in ubuntu20.04 noetic

dbdxnuliba, 2023-05-09:
how to install quad-sdk in ubuntu20.04 noetic

elpimous, 2023-05-09:
Hello, change branch to noetic_devel, or other.

elpimous, 2023-05-09:
In your quad sdk :
Git checkout noetic_devel

dbdxnuliba, 2023-05-16:
thanks @elpimous @selvingarciag96 ,I will have a try later

dbdxnuliba, 2023-05-16:
Errors     << nmpc_controller:make /home/dell/ref/ws_quad_sdk/logs/nmpc_controller/build.make.001.log                                                                                                      
In file included from /home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/nmpc_controller/src/quad_nlp_utils.cpp:1:
/home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/nmpc_controller/include/nmpc_controller/quad_nlp.h:15:10: fatal error: IpIpoptData.hpp: No such file or directory
   15 | #include <IpIpoptData.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nmpc_controller.dir/build.make:328: CMakeFiles/nmpc_controller.dir/src/quad_nlp_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3019: CMakeFiles/nmpc_controller.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

elpimous, 2023-05-16:
@dbdxnuliba
Dis you follow the IPOPT install part?
See quad_sdk install parts...

selvingarciag96, 2023-05-16:
@dbdxnuliba Yes, IPOPT is necessary. Please follow the instructions on https://github.com/robomechanics/quad-sdk/wiki/1.-Getting-Started-with-Quad-SDK. Make sure you are using the correct HSL solver. Let us know if you still have any issues.

dbdxnuliba, 2023-05-17:
ok thanks

closed 2023-05-17 by dbdxnuliba