robomechanics/quad-sdk #370: An issue on the global body planner.

hrxsd, 2023-05-09:
Hello dear developers!  Sorry to bother you again.
I created a map with staircase environment to test the robot's motion planning performance in this SDK, but I'm having trouble with `reference:=gbpl`. Here are the details of my problem:
First, I add the map file to the quadSDK and run `quad_plan.launch`. the robot stops when it reaches the bottom of the stairs and the terminal does not report any error, so it should be that motion planning is continuing but the robot is not moving.
[image]

Then I set `reference:=twist` and the robot can go up the stairs normally. So I think I should be able to rule out the map file itself?
[image]

Next, I modified the _global_body_planner/global_body_planner.yaml_ file to have a `goal_state` of `[1.0, 0.0]`, which is done to keep the motion planning process running normally until it goes up the stairs. After the robot stops, I publish the target point on the stairs using `Publish Point` in Rviz and the terminal reports an error as:
[image]
Of course, the robot is still parked there.

So I would like to know what is causing this problem and how can I fix it? I would be grateful if I could get an answer from you.

selvingarciag96, 2023-05-16:
@hrxsd Thank you for using Quad-SDK ! The current version of the global planner uses heuristics for deciding which motion primitive to apply (Walking or leaping). In this case, the stairs seem to be steep enough that it might not be traversal with walking primitive and the staircase is tall enough that the leap primitive may also be invalid. There are couple solutions to address this:

1. You can play with the heuristics in the global planner to see if that helps. 
2. You can try less steep stairs to identify where the limit is and maybe use that instead
3. If you have a more clever solution to handle such cases, feel free to open a pull request and contribute to Quad-SDK. 

Please let us know if you have any additional questions.

jcnorby, 2023-05-17:
Another note on this - you can play with the [traversability threshold](https://github.com/robomechanics/quad-sdk/blob/b04d4f7658215da018b4843c98f49e50b1b339c6/global_body_planner/global_body_planner.yaml#L13) used by the global planner to determine if the terrain under a reachability point (the corners of the box) are traversable. The planner requires all four corners to be over traversable terrain, so by lowering that threshold (or setting to zero) you can effectively increase the terrain the planner will plan over (although the controller may struggle to track it).

hrxsd, 2023-05-30:
Sorry for not replying in time, as I've been caught up in other things these past two weeks. Just now I tried to set the traversability threshold to 0 and that successfully solved my problem. Thanks again for answering my questions!

closed 2023-05-30 by hrxsd