robomechanics/quad-sdk #371: /usr/local/include/rbdl/rbdl_eigenmath.h:17:1: error: specialization of 'std::vector<Eigen::Matrix<double, 2, 1> >' after instantiation    17 | EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Eigen::Vector2d)

dbdxnuliba, 2023-05-16:
catkin build -j4

Errors     << global_body_planner:make /home/dell/ref/ws_quad_sdk/logs/global_body_planner/build.make.005.log                                                                                              
In file included from /usr/include/eigen3/Eigen/StdVector:23,
                 from /usr/local/include/rbdl/rbdl_math.h:37,
                 from /usr/local/include/rbdl/addons/urdfreader/urdfreader.h:5,
                 from /home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/quad_utils/include/quad_utils/quad_kd.h:5,
                 from /home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/quad_utils/include/quad_utils/math_utils.h:19,
                 from /home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/global_body_planner/include/global_body_planner/planning_utils.h:6,
                 from /home/dell/ref/ws_quad_sdk/src/quad-sdk-noetic_devel/global_body_planner/src/planning_utils.cpp:1:
/usr/local/include/rbdl/rbdl_eigenmath.h:17:1: error: specialization of 'std::vector<Eigen::Matrix<double, 2, 1> >' after instantiation
   17 | EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Eigen::Vector2d)

dbdxnuliba, 2023-05-17:
ok I have fix it now

closed 2023-05-17 by dbdxnuliba

yyagin, 2023-06-15:
Hi, I have the same issue. Could you please share how you fixed it?

TJCberry, 2023-10-14:
@yyagin hello，you can try to Reinstall Eigen library, I reinstalled the version I reinstalled is 3.3.4 ,it worked!