robomechanics/quad-sdk #374: when world use rough_25cm ,the robot rolled over and unable to return to standing

dbdxnuliba, 2023-05-17:
when world use rough_25cm ,the robot rolled over and unable to return to standing  
[image]

ologandavid, 2023-05-23:
Hi @dbdxnuliba, 
At the moment, I double checked the robots performance on the rough_25 cm terrain, and found that it worked fine. See below. Are you running the global planner or a twist input? Also, what branch are you currently running? 

Note that we can't guarantee that the robot will work on all rough terrains, and I'm currently working on a fix for this erratic behavior, but if you would like to issue a pull request and fix this, that'd be great too! 

Best, 
David
[image]

jcnorby, 2023-06-17:
I've actually seen this a few times, and there was a video of it in another issue. The default spawn location on this map is on a slope, and sometimes when the robot stands up its momentum causes it to flip over. @dbdxnuliba can you confirm if this happens upon switching from sit->stand (and not after trying to track a plan)?

If so, fixes could include changing the default starting location slightly to the left so it is on less of a slope, or updating the map in solidworks to be completely flat around the origin.

ardalantj, 2023-06-24:
Checking in on this @dbdxnuliba. Have you been able to resolved the issue by changing the default starting position?

ardalantj, 2023-07-17:
Closing due to inactivity.

closed 2023-07-17 by ardalantj