robomechanics/quad-sdk #376: Running the code in Spirit 40

henryliuliuliu, 2023-05-18:
Hi, I am trying to deploy the quad sdk in Spirit 40. I have two issues which are described below:

I loaded the customized firmware from the RML lab. The suggested way to run SDK is to run "load_robot_env.sh" in TX2/Xavier, and then run "load_remote_env.sh" in the remote controller. 
The first question: Do we have to install all the dependencies for only running the "robot_driver.launch" in the TX2? I used a docker to install the environment, but every time when I set up the code, the problem of installing the ipopt libraries always happens.
I ran the full stack quad sdk in the remote computer:
Below is my procedure:
I first ran: `roslaunch quad_utils robot_driver.launch mocap:=false logging:=true`, then the terminal shows that it successfully connected to the spirit robot with 1000hz, although the frequency is not stable. 
Then I ran, `rostopic pub /robot_1/control/mode std_msgs/UInt8 "data: 1"`
The first time, the robot did not stand as I expected, but I found the controlModeCallback is not called. It turns out that the namespace robot_1 is not loaded correctly. So I manually change the controlModeTopic, which is loaded from the parameter server before to "robot_1/control/mode". Then the above cmd successfully makes the robot stands. 

Do you know the reason?

At last, I ran the remote_driver.launch" in the remote computer with default parameters, it opens the rviz, however, it did not spawn a spirit robot, since it did not call the quad_spawn.launch. I am confused about the goal of opening a rviz panel. I thought it was to show the real-time position of the real spirit. 
[image]

Then I ran: `roslaunch quad_utils quad_plan.launch reference:=twist logging:=true`, in which case, the nmpc will not going to the for loop, I believe that it is blocked by some required information, such as ground force? If so, is there a solution to run the robot without force plate on the ground. 

Thanks for your time! Any explanation would be appreciated.!

henryliuliuliu, 2023-05-18:
Looks like logging.launch will change all topic names to /robot_1/xxxx?

ologandavid, 2023-05-23:
Hi @henryliuliuliu,
I know that you met with Jiming at some point in the week to discuss your setup. I was wondering if you still needed help with this or if I could close the issue. 

Best, 
David

henryliuliuliu, 2023-05-23:
Thanks for your help! I am ok with the quadSDK now!