robomechanics/quad-sdk #377: An error occurred during the initial compilation process

nanbwrn, 2023-05-25:
I encountered an undefined reference error during compilation. The specific content is as follows: Thank you for your reply:
[image]

nanbwrn, 2023-05-27:
Hi @ologandavid ，
thank you for your reply. I am using Ipopt-3.12.7,and HSL for IPOPT as shown in the following figure,
[image]

ologandavid, 2023-05-30:
Hi @nanbwrn,
The Ipopt version that you should be using is 3.13. That's probably contributing to the fact that the initial compliation is failing. Try requesting ma27 or ma57 solver from coinhsl. Let me know if that resolves your issue.

Best, 
David

nanbwrn, 2023-05-31:
Thank you for your enthusiastic response. I have solved this problem and used a new coinhsl that includes ma57 to overwrite the original coinhsl.And it has been compiled successfully

closed 2023-05-31 by nanbwrn