robomechanics/quad-sdk #378: Error reporting when I run roslaunch quad_utils quad_gazebo.launch

nanbwrn, 2023-06-02:
The following error occurred in the terminal when I run roslaunch quad_ utils quad_gazebo. launch.May I ask if anyone has encountered it before? Thank you for your reply
[image]
[image]

ologandavid, 2023-06-12:
Hi @nanbwrn,
Sorry for the late reponse, this looks similar to a problem I resolved in the past. Do you still need help with this, and if so, could I see your rviz settings (display tab), and if possible general information about your system (what branch are you currently using etc.) Checking the gazebo model path in your bash file could also help.

Best, 
David Ologan

closed 2023-06-13 by nanbwrn

pranavgo, 2023-09-11:
@nanbwrn I am facing the same issue. Can you please tell me how you fixed it? Thanks.

nanbwrn, 2023-09-19:
1.I updated Gazebo at that time:
sudo apt upgrade libignition-math2
2.Then, I updated this .sh file, opened the terminal at the quad_simulator directory, and entered: ./setup_deps.sh