robomechanics/quad-sdk #379: What modifications need to be made in the local planner when importing a new robot such as Unitree go1 for simulation

nanbwrn, 2023-06-12:
I generated C++code in Matlab and copied them to nmpc_controller/src/gen and nmpc_controlle/include/nmpc_controlle/gen.
I am a newcomer to quadruped robots, and due to concerns about unexpected errors, I only filled the urdf and sdf of go1 into the relevant sections of the a1 file.During my simulation, I found that the trajectory planned by the planner still seemed to be the same as the original a1, because I found that the trajectory did not seem to coincide with the robot's path. I have reviewed the issues posted by everyone and found that there seems to be a need to modify the basic content:

[image]

Could you please kindly provide me with a detailed explanation of this issue and specify the specific aspects that require modification?
[image]
[image]

[image]

https://github.com/robomechanics/quad-sdk/assets/132453617/03785d99-688b-4675-9c9d-251e719fed95

ologandavid, 2023-06-12:
Hi @nanbwrn,
To start, double check that the parameters of the a1 defined [here] (https://github.com/robomechanics/quad-sdk/blob/main/nmpc_controller/scripts/main.m) are properly approximated. Physically, Spirit is very different from a1 so nailing these values should help. If the issue persists, there are other solutions we can try to get the robot working in sim properly.

Best, 
David Ologan

nanbwrn, 2023-06-13:
@ologandavid 
I have repeatedly checked the parameters of a1 in MATLAB and have indeed changed them to the parameters of go1. May I ask if there are any other programs that need to be changed besides modifying main.m

nanbwrn, 2023-06-13:
The following is my main. m file I modified, and I copied the regenerated C++file to the gen directory.

clear all; clc; close all;

addpath('./utils')
%% Physics parameter
parameter.physics.gravitational_constant=9.81; % Gravity

%parameter.physics.sim2real_scale_factor=(13.3-11.6620+5.75)/5.75; % Real spirit
 parameter.physics.sim2real_scale_factor=1; % Sim spirit or a1

%parameter.physics.mass_body_body=parameter.physics.sim2real_scale_factor*5.75; % Only body weight of spirit
parameter.physics.mass_body_body=parameter.physics.sim2real_scale_factor*5.2; % Only body weight of a1

%parameter.physics.mass_body_leg=1.478; % Each leg weight of spirit
 parameter.physics.mass_body_leg=1.9739; % Each leg weight of a1

parameter.physics.mass_body=parameter.physics.mass_body_body+...
    4*parameter.physics.mass_body_leg; % Total body weight

%parameter.physics.hip_offset=[0.2263; 0.098; 0]; % Absolute hip offset from body COM of spirit
parameter.physics.hip_offset=[0.1881; 0.04675; 0]; % Absolute hip offset from body COM of a1

%parameter.physics.inertia_body=parameter.physics.sim2real_scale_factor*...
   % diag([0.05; 0.1; 0.1]); % Body inertia of spirit
parameter.physics.inertia_body=parameter.physics.sim2real_scale_factor*...
     [0.0168128557, -2.296769e-4, -2.945293e-4;
     -2.296769e-4, 0.063009565, -4.18731e-5;
     -2.945293e-4, -4.18731e-5, 0.0716547275]; % Body inertia of a1

parameter.physics.inertia_body=parameter.physics.inertia_body+...
    4*parameter.physics.mass_body_leg*...
    diag([parameter.physics.hip_offset(2)^2+parameter.physics.hip_offset(3)^2;
    parameter.physics.hip_offset(1)^2+parameter.physics.hip_offset(3)^2;
    parameter.physics.hip_offset(1)^2+parameter.physics.hip_offset(2)^2]); % Robot inertia (assume leg mass concentrated at hip)

parameter.name = "a1"; % Model name
parameter.n = 12; % State dimension
parameter.m = 12; % Input dimension

%% Generate Dynamics Model
dynamicsModel(parameter);

nanbwrn, 2023-06-15:
[go1.xacro.zip](https://github.com/robomechanics/quad-sdk/files/11758169/go1.xacro.zip)

mfclabber, 2023-06-15:
That I do?

Чт, 15 июня 2023 г. в 16:44, Dmitriy Novichkov ***@***.***

nanbwrn, 2023-06-15:
xiangzhang870@gmail

nanbwrn, 2023-06-16:
My code modifications only involved generating the C++ files for the go1 robot from Matlab files in the nmpc_controller, and then copying these C++ files to the gen directory. Are there any other modifications that need to be made? 
Based on the gait below, it gives a sense that although there are no errors, it seems that there is still something that has not been fully modified. The width between the two trajectories generated by Spirit is approximately 7 cells, and the distance between the trajectories generated by the replaced Go1 robot is also 7 cells. This is incorrect.
[image]
[image]

jcnorby, 2023-06-17:
I may be able to help a little bit here - it looks like the difference (or rather, lack of a difference) you are noting is in the planned foothold locations, which you rightly note should be different for different robots. The following logic sets those locations, and following it through to robot parameters may help debug:

- Planned footholds are set in LocalFootstepPlanner::computeFootPlan(), which calls QuadKD::worldToNominalHipFKWorldFrame() to get the location "underneath" the hip. This is the quantity which you are identifying is incorrect.
- QuadKD has internal values of leg lengths which are used to compute these quantities. The parameters in particular that you''ve noted as seeming incorrect are stored g_body_legbases_, which are the transforms from the base_link frame to the abad frames, and in l0_vec_, which is the length from the abad joint to the hip joint (its essentially the y-axis offset of the foot).
- QuadKD loads these quantities in its constructor, using RBDL to load them from a URDF.

Putting these together, I would recommend checking the values of those quantities in QuadKD to see if they are what you expect them to be, and I suspect the issue is in the values in the URDF or in the parsing if the link names don't align. Hope that helps!

nanbwrn, 2023-06-18:
Thank you @jcnorby . I will follow your steps to make the modifications and hope to achieve the desired results.

ardalantj, 2023-06-24:
@nanbwrn Has there been any updates on this?

closed 2023-06-28 by nanbwrn