robomechanics/quad-sdk #380: Help about noetic devel ekf branch

elpimous, 2023-06-16:
@ologandavid,

1 does your actual code works on a real robot?
(ready to test on mine)

2 this ekf version removed the need of mocap or t265, correct?

3 do i need foot sensors, here? (dont have)
Thanks friend.
Vincent.

elpimous, 2023-06-16:
compilation ok, but errors when tests : 

```
roslaunch quad_utils load_global_params.launch load_robot_params:=true
cd ~/catkin_ws
catkin run_tests
```

[image]

elpimous, 2023-06-18:
small up guys ?

ologandavid, 2023-06-19:
Hi @elpimous,
Please hold off for now on testing that branch on your robot. It hasn't been hardware tested and there are still a few bugs I'm currently working through. (1) Yes it should remove the need for mocap or the tracking camera. (2) and no you should't (3).

Best, 
David Ologan

elpimous, 2023-06-19:
Thanks friend...
I'll  wait.

elpimous, 2023-06-19:
@ologandavid ,
Hi friend,
What could be the best branch to rest your controller on my real robot ?
My hardware_interface is ready.
I encounter some errors with melodic_devel_go_outdoors branch (on my noetic system)

elpimous, 2023-07-15:
@ologandavid  please up.

ardalantj, 2023-07-27:
I will close this issue since state estimation this will be part of an upcoming release.

closed 2023-07-27 by ardalantj

dhajnes, 2023-08-03:
Hi @ologandavid, it has been a few weeks since you mentioned the bugs and @ardalantj mentioned that state estimation will be part of an upcoming release. Do you have any release time estimates? In a week? Month? 4 months? I was looking for some roadmap and didn't find any.

ologandavid, 2023-08-03:
[See](https://github.com/robomechanics/quad-sdk/pull/338#issuecomment-1652470683) . Currently testing our implementation on hardware, once its tested, everything should be good to go.

elpimous, 2023-08-03:
Exciting!!!!
Please share sole hardware videos, 😍😍